orbbec/ros_astra_camera #3: Publishing registered point clouds causes extrapolation exceptions

tfoote, 2016-06-01:
```
[ WARN] [1464806618.218487179]: TF2 exception:
Lookup would require extrapolation into the future.  Requested time 1464806618.315158598 but the latest data is at time 1464806618.239545601, when looking up transform from frame [camera_depth_optical_frame] to frame [camera_rgb_optical_frame]
```

This is a static transform so should be easy to work around.

lcluz, 2017-06-08:
I'm having this problem. Is this solved?

ob-tim-liu, 2017-06-09:
@lcluz  could you tell me detailes steps for this problem and OS version ,  hardware Platform ?

lcluz, 2017-06-09:
@ob-tim-liu I'm using a Robot from the project I work in: it uses Ubuntu 14.04 and ROS Indigo. The pointcloud data is good, rgb data is good. Depth registered is getting this extrapolation error and maybe every minute there is some data available.