orbbec/ros_astra_camera #4: Create simple udev rule installation script

tfoote, 2016-06-01:
Kind of like the Kobuki ftdi udev rules installer: https://github.com/yujinrobot/kobuki_core/blob/kinetic/kobuki_ftdi/scripts/create_udev_rules

jonbinney, 2016-06-07:
@tfoote would it be possible to have the debian package install a udev rules file directly into `/etc/udev/rules.d` instead of needing to run a separate script?

tfoote, 2016-06-07:
It's possible, but then we need to worry about colliding files from different distros. And it would need to be a debian specific patch also so we can still support compiling and running without sudo permissions. Since it's a onetime administrative thing I'm not sure building it into the packaging is worth the effort. And the script would be easier to make portable to other platforms too. Such as fedora.

jonbinney, 2016-06-07:
Makes sense, although it would be nice to have that script run automatically on the package install. The current openni packages must do this, right? I don't remember manually running a script to setup the udev rules.

tfoote, 2016-06-07:
We rely on the custom packaged upstream drivers to install the udev rules. They're not installed by the ROS packages.

```
$ dpkg -L libopenni2-0 
/.
/lib
/lib/udev
/lib/udev/rules.d
/lib/udev/rules.d/40-libopenni2-0.rules
/usr
/usr/lib
/usr/lib/libOpenNI2.so.0
/usr/lib/OpenNI2
/usr/lib/OpenNI2/Drivers
/usr/lib/OpenNI2/Drivers/libOniFile.so.0
/usr/lib/OpenNI2/Drivers/libPS1080.so.0
/usr/lib/OpenNI2/Drivers/libPSLink.so.0
/usr/lib/OpenNI2/Drivers/libDummyDevice.so.0
/usr/share
/usr/share/doc
/usr/share/doc/libopenni2-0
/usr/share/doc/libopenni2-0/copyright
/usr/share/doc/libopenni2-0/TODO.Debian
/usr/share/doc/libopenni2-0/changelog.Debian.gz
/etc
/etc/openni2
/etc/openni2/OpenNI.ini
```

closed 2016-06-14 by ob-tim-liu