orbbec/ros_astra_camera #9: Getting "fatal error: libudev.h: No such file or directory  #include <libudev.h>" when compiling the package

Jasmin92, 2016-10-14:
I'm getting "fatal error: libudev.h: No such file or directory  #include <libudev.h>" when compiling the package. This is not probably an issue but I'm guessing that I'm missing a dependency that should be installed.
I would be grateful if you tell me what are the dependencies that should be installed to get the package work properly.
Thanks in advance.

ob-tim-liu, 2016-10-14:
apt-get install libudev-dev

---------------------------------------------------------------------
刘华 
座机：0755-86329228
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区粤兴三道8号中国地质大学产学研基地中地大楼A808
 
From: Jasmin92
Date: 2016-10-14 18:21
To: orbbec/ros_astra_camera
Subject: [orbbec/ros_astra_camera] Getting "fatal error: libudev.h: No such file or directory #include <libudev.h>" when compiling the package (#9)
I'm getting "fatal error: libudev.h: No such file or directory #include " when compiling the package. This is not probably an issue but I'm guessing that I'm missing a dependency that should be installed.
I would be grateful if you tell me what are the dependencies that should be installed to get the package work properly.
Thanks in advance.

Jasmin92, 2016-10-14:
Thank you for your quick answer,
I installed libudev-dev but I'm getting "/usr/bin/ld: cannot find -lOpenNI2Orbbec" is that because the Orbbec Astra SDK is not installed properly?
https://github.com/orbbec/astra
Can you please provide me a step by step instructions in order to install it properly.

ob-tim-liu, 2016-10-14:
No, you need't astra-sdk, ros-astra-camera use OpenNI2 SDK. during catkin_make, you will get OpenNI2 source from https://github.com/tfoote/OpenNI2.git, 
and recompiler it. so be sure to access internet.  
then , you can delete 'devel' and 'build' dir , try it again.

---------------------------------------------------------------------
刘华 
座机：0755-86329228
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区粤兴三道8号中国地质大学产学研基地中地大楼A808
 
From: Jasmin92
Date: 2016-10-14 18:32
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] Getting "fatal error: libudev.h: No such file or directory #include <libudev.h>" when compiling the package (#9)
Thank you for your quick answer,
I installed libudev-dev but I'm getting "/usr/bin/ld: cannot find -lOpenNI2Orbbec" is that because the Orbbec Astra SDK is not installed properly?
https://github.com/orbbec/astra
Can you please provide me a step by step instructions in order to install it properly.

Jasmin92, 2016-10-14:
Thank you very much for these clarifications, it is working fine now.

closed 2016-11-22 by ob-tim-liu