orbbec/ros_astra_camera #10: Certain topics not publishing data

alberto-tellaeche, 2016-11-03:
Hi all,
I have checked driver in ROS Indigo repositories (sudo apt-get install ros-indigo-astra-launch) and also compiling from source.
The driver works, but there are topics that appear doing "rostopic list" that do not publish any data. In my case, these are:

/camera/depth_registered/points
/camera/depth_registered/sw_registered/*
/camera/rgb/*

This means there is not color images, neither RGB-D pointclouds. 

Best,

Alberto

ob-tim-liu, 2016-11-04:
hi
    there are main two type products, astra adn astra pro. i don't know which type you have.
1,astra  using rgb by openni . you can get rgb and points from drivers.
2,astra pro.  with a separate uvc camera, you must get rgb data by your self, such as luvcview.

---------------------------------------------------------------------
刘华 
座机：0755-86329228
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区粤兴三道8号中国地质大学产学研基地中地大楼A808
 
From: alberto-tellaeche
Date: 2016-11-03 20:00
To: orbbec/ros_astra_camera
Subject: [orbbec/ros_astra_camera] Certain topics not publishing data (#10)
Hi all,
I have checked driver in ROS Indigo repositories (sudo apt-get install ros-indigo-astra-launch) and also compiling from source.
The driver works, but there are topics that appear doing "rostopic list" that do not publish any data. In my case, these are:
/camera/depth_registered/points
/camera/depth_registered/sw_registered/*
/camera/rgb/*
This means there is not color images, neither RGB-D pointclouds.
Best,
Alberto

alberto-tellaeche, 2016-11-04:
Hi, thanks for the fast response !
In my case I have an Astra Pro.
I understand your answer, this means ROS driver is not supporting RGB data from the sensors. However, I think it could be extremely helpful to have this colour information from the ROS driver somehow encapsulated, I mean, launch the driver and have all the sensor capabilities in topics, colour included (for example for RGB-D colour and pointcloud registration). Is this planned on the driver development?
If this is not planned, it could be a good idea to remove the topics that are not being used from the list of publishers, just to avoid misunderstandings..

Cheers,

Alberto

ob-tim-liu, 2016-11-04:
you can visit http://wiki.ros.org/uvc_camera,  use this module publish rgb data.
it's goo idea to disable the rgb topic for astra pro.  we will update it next version. thanks!

---------------------------------------------------------------------
刘华 
座机：0755-86329228
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区粤兴三道8号中国地质大学产学研基地中地大楼A808
 
From: alberto-tellaeche
Date: 2016-11-04 16:35
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] Certain topics not publishing data (#10)
Hi, thanks for the fast response !
In my case I have an Astra Pro.
I understand your answer, this means ROS driver is not supporting RGB data from the sensors. However, I think it could be extremely helpful to have this colour information from the ROS driver somehow encapsulated, I mean, launch the driver and have all the sensor capabilities in topics, colour included (for example for RGB-D colour and pointcloud registration). Is this planned on the driver development?
If this is not planned, it could be a good idea to remove the topics that are not being used from the list of publishers, just to avoid misunderstandings..
Cheers,
Alberto

closed 2016-11-22 by ob-tim-liu