orbbec/ros_astra_camera #15: /usr/bin/ld: cannot find -lOpenNI2Orbbec

aswinsarang, 2016-12-20:
@tfoote Receiving the following error.
```
aswin@aswin-laptop:~$ cd camera/
aswin@aswin-laptop:~/camera$ catkin_make 
Base path: /home/aswin/camera
Source space: /home/aswin/camera/src
Build space: /home/aswin/camera/build
Devel space: /home/aswin/camera/devel
Install space: /home/aswin/camera/install
####
#### Running command: "make cmake_check_build_system" in "/home/aswin/camera/build"
####
####
#### Running command: "make -j8 -l8" in "/home/aswin/camera/build"
####
[  3%] [  7%] Built target astra_camera_gencfg
[ 11%] Built target astra_usb_reset
Performing update step for 'astra_openni2'
[ 11%] Built target _astra_camera_generate_messages_check_deps_GetSerial
[ 22%] [ 22%] [ 25%] Built target astra_camera_generate_messages_lisp
Built target astra_camera_generate_messages_py
Built target astra_camera_generate_messages_cpp
[ 25%] Built target astra_camera_generate_messages
Already on 'orbbec_ros'
Your branch is up-to-date with 'origin/orbbec_ros'.
[ 29%] Performing configure step for 'astra_openni2'
no need to configure
[ 33%] Performing build step for 'astra_openni2'
make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
[ 37%] Performing install step for 'astra_openni2'
[ 40%] Completed 'astra_openni2'
[ 51%] Built target astra_openni2
Linking CXX shared library /home/aswin/camera/devel/lib/libastra_wrapper.so
/usr/bin/ld: cannot find -lOpenNI2Orbbec
collect2: error: ld returned 1 exit status
make[2]: *** [/home/aswin/camera/devel/lib/libastra_wrapper.so] Error 1
make[1]: *** [ros_astra_camera/CMakeFiles/astra_wrapper.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

ob-tim-liu, 2017-01-05:
pls rebuild the project

closed 2017-01-05 by ob-tim-liu

Ashkan372, 2017-08-29:
This only happens in this forked repository package, does not happen in the original one
[(https://github.com/orbbec/ros_astra_camera)]