orbbec/ros_astra_camera #32: 1280x960 color mode

chataign, 2017-07-27:
According to the datasheet, the Astra can produce 1280x960 color images, however there is no option in the ROS driver dynamic config for this resolution?

ob-tim-liu, 2017-07-28:
you should set the resolution to 1280x1024, in this condition, you will get 1280x960 with 1280x64 invalid data.

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 1703
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Francois Chataigner
Date: 2017-07-27 19:26
To: orbbec/ros_astra_camera
CC: Subscribed
Subject: [orbbec/ros_astra_camera] 1280x960 color mode (#32)
According to the datasheet, the Astra can produce 1280x960 color images, however there is no option in the ROS driver dynamic config for this resolution?

chataign, 2017-07-31:
thank you, we managed to get HD images using this resolution, however they come at 6.5Hz as opposed to the 10Hz specified in the Astra datasheet

https://orbbec3d.com/product-astra

Is there a way to improve this?

Kind regards,

ob-tim-liu, 2017-08-01:
In the best transfer bandwidth for usb2.0, the frame rate will come to 10fps,
but in normal condition,  we can only  get 7fps. 
we will correct  this description soon.
  

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 1703
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Francois Chataigner
Date: 2017-07-31 17:20
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] 1280x960 color mode (#32)
thank you, we managed to get HD images using this resolution, however they come at 6.5Hz as opposed to the 10Hz specified in the Astra datasheet
https://orbbec3d.com/product-astra
Is there a way to improve this?
Kind regards,

danFromTelAviv, 2018-06-11:
I am using openni( 2.2 ) and I want to get this resolution as well. 
If I set the rgb stream to  1280x1024, 30 fps the function rgb_stream.read_frame() gets stuck and never returns anything. 
If I set rgb stream to 1280x1024 and fps <30 I get an error : ONI_STATUS_NOT_SUPPORTED, b'Stream setProperty(3) failed'
I assume property 3 is fps. 
What should I try ? 

He is the basic code : 
import numpy as np
import cv2
from primesense import openni2  # , nite2
from primesense import _openni2 as c_api
import platform
import inspect
import os

current_dir = os.path.dirname(os.path.abspath(inspect.stack()[0][1]))
distribution = current_dir + "/../OpenNI2/Redist"

openni2.initialize(distribution)
if (openni2.is_initialized()):
     print("openNI2 not initialized")

## Register the device
device = openni2.Device.open_any()

## Create the streams
rgb_stream = device.create_color_stream()

rgb_stream.set_video_mode(
            c_api.OniVideoMode(pixelFormat=c_api.OniPixelFormat.ONI_PIXEL_FORMAT_RGB888,
                               resolutionX=1280,
                               resolutionY=1024, fps=10))

while True:
    bgr = np.fromstring(rgb_stream.read_frame().get_buffer_as_uint8(), dtype=np.uint8).reshape(
        1024, 1280, 3)
    rgb = cv2.cvtColor(bgr, cv2.COLOR_BGR2RGB)
    cv2.imshow("rgb", rgb)
    cv2.waitKey(0)

closed 2019-08-01 by Nan-Orbbec3d-US