orbbec/ros_astra_camera #35: Yellowish RGB image

tkelestemur, 2017-11-14:
Hello, 

We are using Orbbec Astra with our TurtleBots. They are shipped from ClearPath. We get RGB image using `rosrun libuvc_camera camera_node` but it seems to me that the RGB images are not quite good. You can see the RGB image below. Does anybody know how we can fix this?

Note: We have 4 turtlebots and 4 cameras and we get the same results for all of them.

[image]

ob-tim-liu, 2017-11-15:
hi
  from your description above,  you should use the "astra pro".   the color camera of astra pro base on uvc protocol.
yellowwish rgb image indicate you start libuvc _camera without auto_white_balance setting.  you can try to start the 
your device with attachment file.

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Tarik Kelestemur
Date: 2017-11-15 01:13
To: orbbec/ros_astra_camera
CC: Subscribed
Subject: [orbbec/ros_astra_camera] Yellowish RGB image (#35)
Hello,
We are using Orbbec Astra with our TurtleBots. They are shipped from ClearPath. We get RGB image using rosrun libuvc_camera camera_node but it seems to me that the RGB images are not quite good. You can see the RGB image below. Does anybody know how we can fix this?
Note: We have 4 turtlebots and 4 cameras and we get the same results for all of them.

tkelestemur, 2017-11-15:
I don't see the attachment file, can you send it againg?

ob-tim-liu, 2017-11-16:
pls refer http://wiki.ros.org/libuvc_camera 

<launch>
  <group ns="camera">
    <node pkg="libuvc_camera" type="camera_node" name="mycam">
      
      <param name="vendor" value="0x0"/>
      <param name="product" value="0x0"/>
      <param name="serial" value=""/>
      
      <param name="index" value="0"/>

      
      <param name="width" value="640"/>
      <param name="height" value="480"/>
      
      <param name="video_mode" value="uncompressed"/> 
      <param name="frame_rate" value="15"/>

      <param name="timestamp_method" value="start"/> 
      <param name="camera_info_url" value="file:///tmp/cam.yaml"/>

      <param name="auto_exposure" value="3"/> 
      <param name="auto_white_balance" value="false"/>
    </node>
  </group>
</launch>

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Tarik Kelestemur
Date: 2017-11-16 03:36
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] Yellowish RGB image (#35)
I don't see the attachment file, can you send it againg?

closed 2019-06-09 by Nan-Orbbec3d-US