orbbec/ros_astra_camera #39: publish frequency very low on TX2

bonne658, 2017-12-12:
Hello everyone,
When I roslaunch astra_launch astra.launch on TX2, the publishing frequency of /camera/depth/points is lower than 1Hz, while on PC is 29Hz. I have try Nofilter OpenNI version as well as the other. 
Do you have any ideas about it?
Thanks a lot!

ob-tim-liu, 2017-12-12:
if you install the driver by apt-get , pls remove it to try.

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: bonne658
Date: 2017-12-12 18:14
To: orbbec/ros_astra_camera
CC: Subscribed
Subject: [orbbec/ros_astra_camera] publish frequency very low on TX2 (#39)
Hello everyone,
When I roslaunch astra_launch astra.launch on TX2, the publishing frequency of /camera/depth/points is lower than 1Hz, while on PC is 29Hz. I have try Nofilter OpenNI version as well as the other.
Do you have any ideas about it?
Thanks a lot!

bonne658, 2017-12-13:
After removing, the astra_launch package can't be found...

ob-tim-liu, 2017-12-15:
pls refer https://github.com/orbbec/ros_astra_launch

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: bonne658
Date: 2017-12-13 09:55
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] publish frequency very low on TX2 (#39)
After removing, the astra_launch package can't be found...

closed 2019-08-01 by Nan-Orbbec3d-US