orbbec/ros_astra_camera #40: Strange bright pixels on the edge of depth image

Maverobot, 2017-12-14:
Hello,

I have been experiencing a strange issue on my pointcloud, which has a ray (sometimes multiple rays) on the edge of the scene. After checking the depth image from the Astra camera, I found incorrect bright pixels, indicating the gradually increasing depth values, on the edge of the depth image.

Pointcloud: 
[image]

Depth Image:
[image]

ob-tim-liu, 2017-12-15:
how many devices on this condition?  this look like a hardware issues,  pls feed back it on https://3dclub.orbbec3d.com/.
some people will contact you, thx!

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Maverobot
Date: 2017-12-14 17:24
To: orbbec/ros_astra_camera
CC: Subscribed
Subject: [orbbec/ros_astra_camera] Strange bright pixels on the edge of depth image (#40)
Hello,
I have been experiencing a strange issue on my pointcloud, which has a ray (sometimes multiple rays) on the edge of the scene. After checking the depth image from the Astra camera, I found incorrect bright pixels, indicating the gradually increasing depth values, on the edge of the depth image.
Pointcloud:
Depth Image:

Maverobot, 2017-12-15:
I will check all the devices we have once I go to work on next Monday.

Maverobot, 2017-12-18:
I just checked all our three Astra cameras, and they all have the same issue. We bought them separately, and the time interval was even months. Are you sure it's a hardware defect? 

The problem could be solved by removing the white pixels on the depth image in the processing phase. It makes sense to me that this fix happens in the astra_camera node. What do you think of it?

[image]

ob-tim-liu, 2017-12-18:
ok,  you can try to use the filterlibary  on Astra.  it will filter the fix noise.
you can refer the https://github.com/orbbec/ros_astra_camera  readme -DFILTER

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Maverobot
Date: 2017-12-18 17:56
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] Strange bright pixels on the edge of depth image (#40)
I just checked all our three Astra cameras, and they all have the same issue. We bought them separately, and the time interval was even months. Are you sure it's a hardware defect?
The problem could be solved by removing the white pixels on the depth image in the processing phase. It makes sense to me that this fix happens in the astra_camera node. What do you think of it?

Maverobot, 2017-12-18:
I went to https://github.com/orbbec/ros_astra_camera, checked out `filterlibrary` branch, built the ROS package using ` catkin_make --pkg astra_camera -DFILTER=ON`. However, nothing has changed.

ob-tim-liu, 2017-12-19:
visit https://www.dropbox.com/s/e9alg5v9rtq6ue3/GetDebugInfo.zip?dl=0  
then use the tools  get oni file and debuginfo, then send to me , thx!

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Maverobot
Date: 2017-12-18 21:37
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] Strange bright pixels on the edge of depth image (#40)
I went to https://github.com/orbbec/ros_astra_camera, checked out filterlibrary branch, built the ROS package using catkin_make --pkg astra_camera -DFILTER=ON. However, nothing has changed.

Maverobot, 2017-12-19:
I sadly don't have Windows PC setup. Can you please send me a Linux version of debug tool?

ob-tim-liu, 2017-12-19:
you can use below tools save oni , then  send oni file and console log to me .
https://www.dropbox.com/s/0u79ro8t17ze3qk/SaveOni-1219.tgz?dl=0

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Maverobot
Date: 2017-12-19 18:00
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] Strange bright pixels on the edge of depth image (#40)
I sadly don't have Windows PC setup. Can you please send me a Linux version of debug tool?

Maverobot, 2017-12-20:
Here is the log file.

[debug_info.txt](https://github.com/orbbec/ros_astra_camera/files/1575021/debug_info.txt)

Maverobot, 2017-12-22:
Sorry. Here it is.
[capture.oni.tar.gz](https://github.com/orbbec/ros_astra_camera/files/1581954/capture.oni.tar.gz)

ob-tim-liu, 2017-12-22:
device is so close to target ,  no depth show.  can't see the noise 

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Maverobot
Date: 2017-12-22 16:46
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] Strange bright pixels on the edge of depth image (#40)
Sorry. Here it is.
capture.oni.tar.gz

Maverobot, 2018-01-02:
[capture.oni.tar.gz](https://github.com/orbbec/ros_astra_camera/files/1597054/capture.oni.tar.gz)
There is no way for to check whether the recorded depth has the noise. Tell me please if there is no noise to see.

ob-tim-liu, 2018-01-03:
no noise base this recorded file.  i think you can't get filter libray correctly. you can use 'roscd ' confirm the astra_camera and astra_launch  dir , the below link is filter and unfilter oni file record. you can record for compare
https://www.dropbox.com/s/dsycuo8y8m3164a/SaveOni-UnFilter.tgz?dl=0
https://www.dropbox.com/s/zi494trfb9ys674/SaveOni-Filter.tgz?dl=0

Maverobot, 2018-01-03:
Package astra_camera and astra_launch are both at the correct path. This time I made sure the nose were there with this camera pose. 

Here is the recorded debug info.
https://drive.google.com/file/d/14NQ6UXTiYVc9TDS_bbFUY_mobVB9nyhd/view?usp=sharing

Maverobot, 2018-01-18:
Any update on this issue?

Maverobot, 2018-03-19:
@ob-tim-liu  Can you reproduce the issue on your side?

ob-tim-liu, 2018-03-20:
i can't reproduce with use filter library. watch from your last upload,  use filter depth is good , use unfilter there are noises on edge . i think you can use filter library to resolve noise issue.

Maverobot, 2018-04-05:
Sadly not. I have already tried multiple times with filter library, which does not help at all. Furthermore I think my issue is different than what was described [here](https://github.com/orbbec/ros_astra_camera/issues/7). In that issue, the noise happens at the edge of objects while in my case the noise always occurs at the edge of depth image. I consider them as totally different issues.

Some more suggestions for debug would be very welcome.

@tfoote @ob-tim-liu @mintar

Maverobot, 2018-09-06:
Update:

I found out if I launch astra.launch with `depth_registration:=false`, the problem is gone. I can get perfect pointcloud again.