orbbec/ros_astra_camera #41: registered point cloud

duartepolonia, 2017-12-18:
Hello,

I have an Orbbec Persee and I want to obtain a RGB point cloud of different objects.

My operating system is Ubuntu 16.04 LTS and I have ROS Kinetic installed.

I installed this driver and the camera publishes both depth, IR and RGB topics fine (I can see them all in rviz).

However, I cannot get the rgb point cloud!

Could you please suggest me a solution for my problem?

Thank you.

ob-tim-liu, 2017-12-19:
pointclound2 
topic  /camera/depth_registered/points
color transformer  rgb8

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: duartepolonia
Date: 2017-12-18 21:15
To: orbbec/ros_astra_camera
CC: Subscribed
Subject: [orbbec/ros_astra_camera] registered point cloud (#41)
Hello,
I have an Orbbec Persee and I want to obtain a RGB point cloud of different objects.
My operating system is Ubuntu 16.04 LTS and I have ROS Kinetic installed.
I installed this driver and the camera publishes both depth, IR and RGB topics fine (I can see them all in rviz).
However, I cannot get the rgb point cloud!
Could you please suggest me a solution for my problem?
Thank you.

duartepolonia, 2017-12-19:
ob-tim-liu, I didn't understand your reply.
I cannot see topic  /camera/depth_registered/points, there are no messages received.

Basically, the node that should be publishing this topic /camera/depth_registered/points is not being subscribed. I will explain the issue again, to be more clear:

-I want to get a xyzrgb pointcloud using a Orbbec Persee camera
-I installed this driver: https://github.com/clearpathrobotics/ros_astra_launch
-The drivers runs well except for getting published the /camera/depth_registered/points topic (no messages received)
-Diving deep into the problem, I explored the driver and found out that the node that should be publising this topic is: depth_image_proc/point_cloud_xyzrgb, but it is not being subscribed and therefore it does not publish anything

Does anyone experienced the same problem? I appreciate any help to solve this issue.

Thank you.

ob-tim-liu, 2017-12-19:
sorry, the  RGB Camera of Orbbec Persee  is uvc camera, so you can't use astra_camera node  to get rgb data.
we will publish rgb data for Persee and Astra Pro  in next release version. 

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: duartepolonia
Date: 2017-12-19 16:42
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] registered point cloud (#41)
ob-tim-liu, I didn't understand your reply.
I cannot see topic /camera/depth_registered/points, there are no messages received.
Basically, the node that should be publishing this topic /camera/depth_registered/points is not being subscribed. I will explain the issue again, to be more clear:
-I want to get a xyzrgb pointcloud using a Orbbec Persee camera
-I installed this driver: https://github.com/clearpathrobotics/ros_astra_launch
-The drivers runs well except for getting published the /camera/depth_registered/points topic (no messages received)
-Diving deep into the problem, I explored the driver and found out that the node that should be publising this topic is: depth_image_proc/point_cloud_xyzrgb, but it is not being subscribed and therefore it does not publish anything
Does anyone experienced the same problem? I appreciate any help to solve this issue.
Thank you.

duartepolonia, 2017-12-19:
Sorry my double comment, this last was duplicated by mistake.

Anyway, the driver that I mentioned above uses libuvc to get the RGB data and it works fine. The only problem the the registration part, I cannot get the /camera/depth_registered/points topic with data!!. The RGB and Depth cameras are not synchronized, and this may be the problem?

Bellow the libuvc part of the driver:

<group ns="$(arg rgb)" if="false">

      <node pkg="libuvc_camera" type="camera_node" name="astra_pro_uvc">
        
        <param name="vendor" value="0x2bc5"/>
        <param name="product" value="0x0501"/>
        <param name="serial" value=""/>
        
        <param name="index" value="0"/>

        
        <param name="width" value="640"/>
        <param name="height" value="480"/>
        
        <param name="video_mode" value="uncompressed"/> 
        <param name="frame_rate" value="15"/>
        <param name="frame_id" value="$(arg camera)_rgb_optical_frame"/>

        <param name="timestamp_method" value="start"/> 
        

        <param name="auto_exposure" value="3"/> 
        <param name="auto_white_balance" value="false"/>
      </node>

    </group> 

  </group>

ob-tim-liu, 2017-12-19:
i know,  the register topic subscribe by rgbd_launch,  you can refer the   depth_registered.launch.xml  in rgbd_launch/launch

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: duartepolonia
Date: 2017-12-19 18:56
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] registered point cloud (#41)
Sorry my double comment, this last was duplicated by mistake.
Anyway, the driver that I mentioned above uses libuvc to get the RGB data and it works fine. The only problem the the registration part, I cannot get the /camera/depth_registered/points topic with data!!. The RGB and Depth cameras are not synchronized, and this may be the problem?
Bellow the libuvc part of the driver:
  <node pkg="libuvc_camera" type="camera_node" name="astra_pro_uvc">
    
    <param name="vendor" value="0x2bc5"/>
    <param name="product" value="0x0501"/>
    <param name="serial" value=""/>
    
    <param name="index" value="0"/>
    
    <param name="width" value="640"/>
    <param name="height" value="480"/>
    
    <param name="video_mode" value="uncompressed"/> 
    <param name="frame_rate" value="15"/>
    <param name="frame_id" value="$(arg camera)_rgb_optical_frame"/>
    <param name="timestamp_method" value="start"/> 
    
    <param name="auto_exposure" value="3"/> 
    <param name="auto_white_balance" value="false"/>
  </node>
</group>

duartepolonia, 2017-12-19:
Ok. But the situation is the following:

depth_registered.launch.xml is refered in the driver but still I cannot get the depth_image_proc nodelet working (this is the nodelet that publishes the depth_registered/points topic)

if I do $rostopic hz /camera/rgb/image_raw I get:
    subscribed to [/camera/rgb/image_raw]
    average rate: 12.774
	    min: 0.078s max: 0.078s std dev: 0.00012s window: 11

if I do $rostopic hz /camera/depth/image_raw I get:
    subscribed to [/camera/depth/image_raw]
    average rate: 19.158
	    min: 0.052s max: 0.052s std dev: 0.00008s window: 15

Which means that both cameras are desynchronized!

Also, the  depth_registered/image_rect topic (remapped to depth_registered(sw_registered/image_rect_raw), that is published by the depth_image_proc/register nodelet only publishes data at random time, e.g. 
    $rostopic hz /camera/depth_registered/sw_registered/image_rect_raw
    subscribed to [/camera/depth_registered/sw_registered/image_rect_raw]
    no new messages
    no new messages
    no new messages
    no new messages
    no new messages
    no new messages
    no new messages
    average rate: 0.165
	    min: 6.056s max: 6.056s std dev: 0.00000s window: 2
    no new messages
    no new messages
    no new messages
    no new messages
    no new messages
    average rate: 0.165
	    min: 6.056s max: 6.059s std dev: 0.00107s window: 3

So I assume the problem is about synchronism of the IR and RGB cameras!

I tried to align the timeframe of the RGB camera (through the libuvc driver) with the timeframe IR but it gives me error everytime I change it.

Shouldn't the cameras be synchronized by default? How can I do it?

Thank you

ob-tim-liu, 2017-12-20:
The pclock of  RGB Sensor and IR Sensor had been synchronized  on hardware design, so the timing of exposure is same for both sensor,
but rgb packaged by UVC , ir packaged by OpenNI,  in theory ,  there are 1-2 frame different between RGB and IR.

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: duartepolonia
Date: 2017-12-19 20:11
To: orbbec/ros_astra_camera
CC: ob-tim-liu; Comment
Subject: Re: [orbbec/ros_astra_camera] registered point cloud (#41)
Ok. But the situation is the following:
depth_registered.launch.xml is refered in the driver but still I cannot get the depth_image_proc nodelet working (this is the nodelet that publishes the depth_registered/points topic)
if I do $rostopic hz /camera/rgb/image_raw I get:
subscribed to [/camera/rgb/image_raw]
average rate: 12.774
min: 0.078s max: 0.078s std dev: 0.00012s window: 11
if I do $rostopic hz /camera/depth/image_raw I get:
subscribed to [/camera/depth/image_raw]
average rate: 19.158
min: 0.052s max: 0.052s std dev: 0.00008s window: 15
Which means that both cameras are desynchronized!
Also, the depth_registered/image_rect topic (remapped to depth_registered(sw_registered/image_rect_raw), that is published by the depth_image_proc/register nodelet only publishes data at random time, e.g.
$rostopic hz /camera/depth_registered/sw_registered/image_rect_raw
subscribed to [/camera/depth_registered/sw_registered/image_rect_raw]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
average rate: 0.165
min: 6.056s max: 6.056s std dev: 0.00000s window: 2
no new messages
no new messages
no new messages
no new messages
no new messages
average rate: 0.165
min: 6.056s max: 6.059s std dev: 0.00107s window: 3
So I assume the problem is about synchronism of the IR and RGB cameras!
I tried to align the timeframe of the RGB camera (through the libuvc driver) with the timeframe IR but it gives me error everytime I change it.
Shouldn't the cameras be synchronized by default? How can I do it?
Thank you

closed 2018-03-16 by ob-tim-liu