orbbec/ros_astra_camera #45: why cann't i use ApproximateTimeSynchronizer or ExactTimeSynchronizer to subscribe rgb and depth topics simultaneously?

pkr97, 2018-03-15:
I want to subscribe two topics at a time using ApproximateTimeSynchronizer for Astra Pro camera. 

This is what I'm doing to reach my goal.

int main(int argc, char** argv)
{
	ros::init(argc, argv, "hit_detection");

	ros::NodeHandle nh;
	
	// topic subscription
	message_filters::Subscriber<Image> RGB_sub(nh, "/camera/rgb/image", 1);
	message_filters::Subscriber<Image> DEPTH_sub(nh, "/camera/depth/image", 1);
	
	// synchronization policy
	typedef sync_policies::ApproximateTime<Image,Image> MySyncPolicy;
	
	// ApproximateTime takes a queue size as its constructor argument, hence MySyncPolicy(10)
	Synchronizer<MySyncPolicy> sync(MySyncPolicy(1), RGB_sub, DEPTH_sub);
	sync.registerCallback(boost::bind(&callback, _1, _2));

	ros::spin();
	return 0;
}

and then in callback function I'm doing this...

void callback(const ImageConstPtr& image_rgb, const ImageConstPtr& image_depth)
{

// do something here.

}

when I subscribe one topic only it works but when I'm trying to subscribe using timeSynchronizer it fails. It's not going to the callback. I mean the subscription process is not working well somewhere.

It would be a great help if anyone can solve this issue.

Thanks,
Prasanna

ob-tim-liu, 2018-03-16:
Astra Pro use a UVC camera， this driver don't support to output the rgb topic for Astra Pro. so you must use other driver for RGB of Astra Pro, like libuvc_camera.

pkr97, 2018-03-16:
Hello Liu, I don't know if it was because UVC camera. because this worked nicely in Ubuntu14.04 + ROS Indigo. I tried a lot to use this package in ROS indigo but couldn't. In addition to the sunchronization problem I was unable to get /camera/depth_registered/points. All these problems vanished with ROS indigo.

Cheers,
Prasanna

closed 2018-03-16 by pkr97