orbbec/ros_astra_camera #47: Is it possible to use IR, depth and RGB at 60HZ

pkr97, 2018-03-18:
Hello All,
        I was wondering if I could use the Astra pro at a rate of 60Hz. Currently rgb and depth both are running at 30Hz. I'm currently using 640x480 size frames. But I don't mind using the 320x240 sized frames in order to get 60Hz as is mentioned in Astra.cfg.

output_mode_enum = gen.enum([ gen.const( "SXGA_30Hz", int_t, 1, "1280x1024@30Hz"), 
                                    gen.const( "SXGA_15Hz", int_t, 2, "1280x1024@15Hz"), 
                                    gen.const( "XGA_30Hz", int_t, 3, "1280x720@30Hz"), 
                                    gen.const( "XGA_15Hz", int_t, 4, "1280x720@15Hz"), 
                                    gen.const( "VGA_30Hz", int_t, 5, "640x480@30Hz"), 
                                    gen.const( "VGA_25Hz", int_t, 6, "640x480@25Hz"), 
                                    gen.const( "QVGA_25Hz", int_t, 7, "320x240@25Hz"), 
                                    gen.const( "QVGA_30Hz", int_t, 8, "320x240@30Hz"), 
                                    gen.const( "QVGA_60Hz", int_t, 9, "320x240@60Hz"), 
                                    gen.const( "QQVGA_25Hz", int_t, 10, "160x120@25Hz"), 
                                    gen.const( "QQVGA_30Hz", int_t, 11, "160x120@30Hz"), 
                                    gen.const( "QQVGA_60Hz", int_t, 12, "160x120@60Hz")], "output mode") 

I belive that I need change some configs like 

gen.add("ir_mode", int_t, 0, "Video mode for IR camera", 5, 1, 12, 9, edit_method = output_mode_enum)
gen.add("color_mode", int_t, 0, "Video mode for color camera", 5, 1, 12, 9, edit_method = output_mode_enum)
gen.add("depth_mode", int_t, 0, "Video mode for depth camera", 5, 1, 12, 9, edit_method = output_mode_enum)

and If I could change so that I can get a 60Hz rate for IR, RGB and Depth output.

So, my concern is that, is it possible to do like this and what else I need to achive my goal of getting 60Hz rate for all outputs?

What are the things I need to change ?

I'm using Ubuntu 14.04 and ROS Indigo.

Thanks,
Prasanna

ob-tim-liu, 2018-03-19:
sorry,  rgb of astra pro can't been supported by this driver.  and 60HZ can't been supported by hardware. 

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: pkr97
Date: 2018-03-18 22:45
To: orbbec/ros_astra_camera
CC: Subscribed
Subject: [orbbec/ros_astra_camera] Is it possible to use IR, depth and RGB at 60HZ (#47)
Hello All,
I was wondering if I could use the Astra pro at a rate of 60Hz. Currently rgb and depth both are running at 30Hz. I'm currently using 640x480 size frames. But I don't mind using the 320x240 sized frames in order to get 60Hz as is mentioned in Astra.cfg.
output_mode_enum = gen.enum([ gen.const( "SXGA_30Hz", int_t, 1, "1280x1024@30Hz"),
gen.const( "SXGA_15Hz", int_t, 2, "1280x1024@15Hz"),
gen.const( "XGA_30Hz", int_t, 3, "1280x720@30Hz"),
gen.const( "XGA_15Hz", int_t, 4, "1280x720@15Hz"),
gen.const( "VGA_30Hz", int_t, 5, "640x480@30Hz"),
gen.const( "VGA_25Hz", int_t, 6, "640x480@25Hz"),
gen.const( "QVGA_25Hz", int_t, 7, "320x240@25Hz"),
gen.const( "QVGA_30Hz", int_t, 8, "320x240@30Hz"),
gen.const( "QVGA_60Hz", int_t, 9, "320x240@60Hz"),
gen.const( "QQVGA_25Hz", int_t, 10, "160x120@25Hz"),
gen.const( "QQVGA_30Hz", int_t, 11, "160x120@30Hz"),
gen.const( "QQVGA_60Hz", int_t, 12, "160x120@60Hz")], "output mode")
I belive that I need change some configs like
gen.add("ir_mode", int_t, 0, "Video mode for IR camera", 5, 1, 12, 9, edit_method = output_mode_enum)
gen.add("color_mode", int_t, 0, "Video mode for color camera", 5, 1, 12, 9, edit_method = output_mode_enum)
gen.add("depth_mode", int_t, 0, "Video mode for depth camera", 5, 1, 12, 9, edit_method = output_mode_enum)
and If I could change so that I can get a 60Hz rate for IR, RGB and Depth output.
So, my concern is that, is it possible to do like this and what else I need to achive my goal of getting 60Hz rate for all outputs?
What are the things I need to change ?
I'm using Ubuntu 14.04 and ROS Indigo.
Thanks,
Prasanna

pkr97, 2018-03-25:
Ok. Tim Thanks for the reply. But these is another aspect that I would like explore and that is how can I use both RGB and Depth at a rate of 30Hz but with a frame size of 320x240.

could you please guide me on this? 

I'm using Ubuntu 14.04, ROS Indigo, and ROS driver for Orbbec Astra Pro.

Thanks,
Prasanna

ob-tim-liu, 2018-03-26:
for depth you can gen.add("depth_mode", int_t, 0, "Video mode for depth camera", 8, 1, 12, 9, edit_method = output_mode_enum), for color , you can refer http://wiki.ros.org/libuvc_camera

pkr97, 2018-03-26:
Thanks Tim for quick reply. I'm facing another problem. I'm using Astra Pro for an application. I ran it for nearly 20 hrs continuously and stopped it after that. Now when I try to run my application I'm getting an error. That error is shown because the RGB camera is up and depth camera is down.

what could be the reason? is there any other way than re-plugging? because I'm using that remotely with an IOT module.

Thanks,
Prasanna

ob-tim-liu, 2018-03-26:
i do't know how do you open rgb camera, and the specific situation of error. For pro rgb and depth is independent, open rgb don't lead to depth faile. i think there are some error log on this condition , you can follow it

pkr97, 2018-03-28:
Hello Tim,
         that problem solved with a simple re-plugging.

anyway I have a new problem.

I'm using Astra SDK for Astra Pro. and using Ubuntu14.04. when trying to compile a .cpp file that I have created for testing, I'm getting this error. 
"
fatal error: astra/astra.hpp: No such file or directory
#include 
^
compilation terminated.
"

I have put the lines below in .bashrc file

"
export ASTRA_SDK_INCLUDE=/home/pkumars/Astra-SDK-Linux/install/include
export ASTRA_SDK_LIB=/home/pkumars/Astra-SDK-Linux/install/lib

"

*** and my CMakeLists.txt looks like this.

"
cmake_minimum_required(VERSION 3.2)

project( AstraTest )

include_directories ("$ENV{ASTRA_SDK_INCLUDE}")

add_executable( test1 main.cpp )

target_link_libraries(test1
${astra_LIBRARIES})
"

hope I made my point clear. any kind of help will be highly appreciated.

Thanks,
Prasanna

closed 2019-08-01 by Nan-Orbbec3d-US