orbbec/ros_astra_camera #49: Setting rgb_preferred to SXGA_30Hz results in an "scrolling image"

mmoerdijk, 2018-04-10:
Setting the rgb_preferred parameter in the Astra.cfg file to 1 results in an image with the correct size, but the image scrolls. I think this has something to do with the maximum height resolution of the camera being 960. but the image is pretty much unusable. I (understand)[https://github.com/orbbec/ros_astra_camera/issues/32] that the intended behaviour would be that a fixed part of the image would have invalid data.

Further more I did some experimenting with the different colour modes and I cannot say I understand which modes are compatible. Is there some documentation on that?

ob-tim-liu, 2018-04-11:
you should contact the seller to upgrade the firmware. pls visit orbbec3d.com for support.

---------------------------------------------------------------------
刘华 
座机：0755-86329228 转 8173
邮箱：liuhua@orbbec.com
手机：13603061506
深圳奥比中光科技有限公司 www.orbbec.com.cn 
地址：深圳市南山区学府路63号高新区联合总部大厦12楼
 
From: Mart Moerdijk
Date: 2018-04-10 23:18
To: orbbec/ros_astra_camera
CC: Subscribed
Subject: [orbbec/ros_astra_camera] Setting rgb_preferred to SXGA_30Hz results in an "scrolling image" (#49)
Setting the rgb_preferred parameter in the Astra.cfg file to 1 results in an image with the correct size, but the image scrolls. I think this has something to do with the maximum height resolution of the camera being 960. but the image is pretty much unusable. I (understand)[https://github.com/orbbec/ros_astra_camera/issues/32] that the intended behaviour would be that a fixed part of the image would have invalid data.
Further more I did some experimenting with the different colour modes and I cannot say I understand which modes are compatible. Is there some documentation on that?

mmoerdijk, 2018-06-22:
Hi,

I tryied to find the update tool but I cannot find it. The topics I could find are all outdated. Could you provide me with the correct tool?

Best Regards,

Mart

These links don’t work
[1] https://3dclub.orbbec3d.com/t/astra-s-firmware-verification-and-update/
[2] https://3dclub.orbbec3d.com/t/rgb-d-sync-project/265

邮箱：liuhua@orbbec.com<mailto:liuhua@orbbec.com>

closed 2019-08-01 by Nan-Orbbec3d-US