orbbec/ros_astra_camera #50: Issue with IR sensor intrinsics at different resolutions (bug)

coxep, 2018-04-13:
I have observed that when running the IR camera in VGA (IR mode 5), the resolution is 640x480. The default camera intrinsics at this resolution seem reasonable to me (image center is at the center of the image, and the focal length calculation in astra_device looks correct)

When I switch to XVGA (IR mode 1), I observe an additional section of pixels (64 pixels high) at the bottom of the IR image with extra information (i.e., these pixels are also being captured by the CMOS sensor, but not included in the VGA image). This suggests that the image is being trimmed at VGA resolution. Do the camera intrinsics account for this?

If they _do_ account for this, then the default intrinsics for XVGA **must** be wrong, since the sensor and field of view do not change, yet the height of the image changes asymmetrically! This results in the focal length at XVGA not being twice the focal length at VGA (as it should, provided that the default intrinsics are correct at VGA resolution)

Clearly, this is a bug with the default intrinsics calculations (either for VGA, XVGA, or both)

Could some clarity be provided concerning the camera sensor? Specifically, I need to know the following:

1) Is the IR stream reporting the vertical field of view of the CMOS sensor, or is it reporting a modified vertical field of view that accounts for the trimmed VGA image?

2) Is the VGA image trimmed symmetrically? (i.e., XVGA image indicates that there is at least 32 pixels of information not included in the bottom of the VGA image. Is there also a 32-pixel section removed from the top of the VGA image?) The image center being reported in both VGA and XVGA modes indicates this to be the case, but I wanted to verify.

Once I know this information, I can issue a pull request for astra_camera to correctly report the resolution at VGA/XVGA

coxep, 2018-04-13:
Also, I examined the fields of view being reported (vertical fov is being used in default intrinsics calcs)

Regardless of IR mode, the vertical field of view is reported as 0.796618 radians. Horizontal field of view is 1.0226 radians. Provided that pixels are square (default intrinsics suggest this is expected), then the following should be true:

tan(vFOV/2) / tan(hFOV) = 0.75

In reality, the value is about 0.78, suggesting that either the focal lengths in x and y are different (pixels are not square) or that the FOVs are incorrect

closed 2019-08-01 by Nan-Orbbec3d-US

coxep, 2019-08-12:
Curious -- what was the resolution to this?

Nan-Orbbec3d-US, 2019-08-12:
XVGA is 1024x768