orbbec/ros_astra_camera #53: Support for Extended API (with IR-Flood)

NikolasE, 2018-04-25:
We are using Mini S cameras (with IR-Flood). So far, we couldn't find the command to toggle the projector and IR-Flood (is this even possible without the Extended API?)

Without this feature, the IR-Flood is quite useless.

NikolasE, 2018-04-30:
I managed to get the IR-Flood working (linked against extended API and then used the same code you used for the flood with a different command). Was the OpenNI_2.3.0.43/ExtendedAPI (with the XN_MODULE_PROPERTY_IRFLOOD_STATE = 0x1080FF94) already officially published?

NikolasE, 2018-05-08:
This CMakelists can be used to link with the Extended API (which can be found in the Download thread at https://3dclub.orbbec3d.com/t/universal-download-thread-for-astra-series-cameras/622):

```
cmake_minimum_required(VERSION 2.8.3)
project(astra_camera)

find_package(catkin REQUIRED camera_info_manager dynamic_reconfigure image_transport  nodelet sensor_msgs roscpp message_generation)

find_package(Boost REQUIRED COMPONENTS system thread)

generate_dynamic_reconfigure_options(cfg/Astra.cfg)
add_service_files(FILES
  GetSerial.srv)
generate_messages()

catkin_package(
  INCLUDE_DIRS include
  LIBRARIES astra_wrapper
  CATKIN_DEPENDS camera_info_manager dynamic_reconfigure image_transport nodelet sensor_msgs roscpp message_runtime
#  DEPENDS libastra
)

# Fetching ExtendedAPI from dropbox:
set(extended_api_base ${CMAKE_CURRENT_BINARY_DIR}/astra_extended_API/src/astra_extended_api/OpenNI-Extension/linux-x64/ExtendedAPI/3rdparty/OpenNI2)
set(extended_api_include ${extended_api_base}/Include)
include(ExternalProject)
ExternalProject_Add(astra_extended_api
        PREFIX astra_extended_API
        TMP_DIR /tmp
        URL  https://www.dropbox.com/sh/bgua92ffs7qdr7r/AADE7ObKGqy8itVCF5EfNVTYa/OpenNI-Extension-2.3.0.43.zip
        URL_HASH MD5=175df83ca31bbe76d210c93df84e7989
        CONFIGURE_COMMAND ""
        BUILD_COMMAND ""
        # these to files are expected to be in openni2/
        INSTALL_COMMAND mkdir -p ${extended_api_include}/openni2  && cp ${extended_api_include}/OpenNI.h ${extended_api_include}/PS1080.h ${extended_api_include}/openni2
)

link_directories(${extended_api_base}/Redist)

include_directories(include
                    ${catkin_INCLUDE_DIRS}
                    ${Boost_INCLUDE_DIRS}
                    ${extended_api_include}
                    )

add_library(astra_wrapper
   src/astra_convert.cpp
   src/astra_device.cpp
   src/astra_device_info.cpp
   src/astra_timer_filter.cpp
   src/astra_frame_listener.cpp
   src/astra_device_manager.cpp
   src/astra_exception.cpp
   src/astra_video_mode.cpp
)
target_link_libraries(astra_wrapper ${catkin_LIBRARIES} -lOpenNI2 ${Boost_LIBRARIES} )

add_dependencies(astra_wrapper astra_extended_api ${PROJECT_NAME}_generate_messages_cpp)

add_library(astra_driver_lib
                src/astra_driver.cpp
)
target_link_libraries(astra_driver_lib astra_wrapper ${catkin_LIBRARIES} ${Boost_LIBRARIES} )
add_dependencies(astra_driver_lib astra_openni2 ${PROJECT_NAME}_gencfg ${PROJECT_NAME}_generate_messages_cpp)

add_library(astra_camera_nodelet
   ros/astra_camera_nodelet.cpp
)
target_link_libraries(astra_camera_nodelet astra_driver_lib ${catkin_LIBRARIES} ${Boost_LIBRARIES} )
add_dependencies(astra_camera_nodelet astra_openni2 ${PROJECT_NAME}_gencfg)

add_executable(astra_camera_node
   ros/astra_camera_node.cpp
)
target_link_libraries(astra_camera_node astra_driver_lib ${catkin_LIBRARIES} ${Boost_LIBRARIES} )
add_dependencies(astra_camera_node astra_openni2 ${PROJECT_NAME}_gencfg ${PROJECT_NAME}_generate_messages_cpp)

add_executable(astra_list_devices
   src/list_devices.cpp
)
target_link_libraries(astra_list_devices astra_wrapper)

add_executable(astra_test_wrapper test/test_wrapper.cpp )
target_link_libraries(astra_test_wrapper astra_wrapper ${Boost_LIBRARIES})
if (UNIX AND NOT APPLE)
  add_executable(astra_usb_reset src/usb_reset.c)
  set(ADDITIONAL_EXECUTABLES "astra_usb_reset")
endif()

install(TARGETS astra_wrapper astra_camera_nodelet astra_camera_node astra_list_devices astra_driver_lib ${ADDITIONAL_EXECUTABLES}
  ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
  )

install(FILES ${extended_api_base}/Redist/libOpenNI2.so
  DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}/
)

install(DIRECTORY ${extended_api_base}/Redist/OpenNI2
  DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}/
)

install(DIRECTORY include/${PROJECT_NAME}/
  DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION}
)

# add xml file
install(FILES astra_nodelets.xml
        DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}
)
```

Toggling for IR can be done as in the code from @coxep with the XN_MODULE_PROPERTY_IRFLOOD_STATE Command in PS1080.h

NikolasE, 2018-11-21:
Any updates on official support?

Nan-Orbbec3d-US, 2019-06-09:
Hi @NikolasE, I integrated the extended API in the latest version. After launching, you can set ir flood by using the rosservice /camera/set_ir_flood to set up ir flood.

Nan-Orbbec3d-US, 2019-07-01:
@NikolasE if you couldn't use this service successfully, please reopen this issue.

closed 2019-07-01 by Nan-Orbbec3d-US