orbbec/ros_astra_camera #55: How do I compile this?

dhiegomaga, 2018-04-27:
I'm sorry this might be a very novice questions for most people, but how do I set up those folders to compile? The Readme doesn't specify anything, if I just open the terminal and type `catkin_make --pkg astra_camera` it obviously won't work. 

Do I just place **eveything** inside _~/catkin_ws/src/ros_astra_camera_ (or _~/catkin_ws/src/astra_camera_? it seems the name of the project is **ros_astra_camera** but the first command to compile the package is **astra_camera**). Do I also need to install any drivers? How? 

Thank you

parthc-rob, 2018-05-31:
> Do I just place eveything inside ~/catkin_ws/src/ros_astra_camera and compile using the catkin_make --pkg astra_camera (or would I put it inside ~/catkin_ws/src/astra_camera?

Yes.

navigate to your `catkin_ws` folder and 

> open the terminal and type catkin_make --pkg astra_camera

If you look into `package.xml` , the package name is astra_camera. The name of the folder you put it in doesn't matter.

Isha8, 2018-07-23:
Were you able to successfully run this package with astra_launch? After running the catkin_make command, when I run roscd astra_camera, it says "roscd: No such package/stack 'astra_camera'".  What could I be doing wrong? Thanks

Isha8, 2018-07-23:
Just worked. Thanks @parthc-rob .

closed 2019-06-09 by Nan-Orbbec3d-US