orbbec/ros_astra_camera #58: Cannot change RGB exposure on Astra Mini S

coxep, 2018-05-18:
There is a reconfigurable option in the ROS driver to toggle "auto_exposure". However, toggling this option does not change anything for the Astra Mini S (the RGB image's exposure still changes automatically with changes to light conditions.) How can I actually disable the auto exposure on the Mini S?

saikishor, 2019-05-02:
@coxep Have you figured it out?

Nan-Orbbec3d-US, 2019-07-10:
Hi @coxep and @saikishor , the latest version provides an rosservice to modify the RGB exposure by the following command.
```sh
rosservice call /camera/set_uvc_exposure "exposure: 100"
```
You can give it a try.

mehditlili, 2019-08-06:
I think it is caused by an older firmware on the camera, I had this problem too but when I asked orbbec for the latest firmware (Can only be uploadexd via Windows) they've sent me an updater and after that I could see changes when I set the exposure.

Nan-Orbbec3d-US, 2019-08-06:
@mehditlili Sounds good! @coxep @saikishor Does anyone still have this issue? Or I will close it.

mehditlili, 2019-08-06:
@Nan-Orbbec3d-US you can't close that issue without providing a link to a downloadable updater. Having to contact customer support to get the updater is not a "fixed" state !
And btw the latest firmware version slightly changes the camera intrinsics and results in different pointclouds, just letting other know.

Nan-Orbbec3d-US, 2019-08-06:
In my opinion, based on your comment, it's not an issue of this package. It's a problem of the firmware. It would be better to have a firmware updater on Orbbec official website, but it's weird that I just post the link here. (Even users not using ROS should encounter this!) Thus, I will write an email to notify them this issue and tell you guys how will it be dealt.