orbbec/ros_astra_camera #60: Failed to load nodelet [/camera/driver] of type [astra_camera/AstraDriverNodelet

suntchan, 2018-06-08:
Hello!

 `roslaunch astra_launch astra.launch` gives me the following error :

`[ERROR] [1528466499.738426497]: Failed to load nodelet [/camera/driver] of type [astra_camera/AstraDriverNodelet] even after refreshing the cache: Could not find library corresponding to plugin astra_camera/AstraDriverNodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
`
Can you help ? I have no idea what to do.
Thank you!

Numjerlunker, 2018-12-12:
I too am struggling with this, did you manage to solve it in the end?

CoryRamsey, 2019-02-23:
I'm also having this issue. Anyone have any luck?

Nan-Orbbec3d-US, 2019-06-09:
Hi folks, could you provide more detailed information (platform, processor, etc.)? Or could you try the latest version again? Thanks!

gstavrinos, 2019-06-26:
I had the same issue in ROS Melodic with the version from the Ubuntu ROS repositories. Cloning the latest version from your Github, fixed the problem.

closed 2019-06-26 by Nan-Orbbec3d-US

Sofiap91, 2020-03-03:
I am having the same issue, I have clone the github repository but still it does not work.
Any ideas?

Saswati-Project, 2024-07-09:
Is there any solution of this problem? I have the same issue. I followed [this](http://clearpathrobotics.com/assets/guides/kinetic/turtlebot/AstraConfig.html) to configure Astra_camera.