orbbec/ros_astra_camera #61: Not working with ROS Melodic

DonSlaughter, 2018-06-12:
I get the following error messages when I try to compile ros_astra_camera :

`[ 40%] Performing configure step for 'astra_openni2'
no need to configure
[ 43%] Performing build step for 'astra_openni2'
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
ThirdParty/PSCommon/BuildSystem/CommonDefs.mak:40: HOST_PLATFORM is x64
../../BuildSystem/CommonDefs.mak:40: HOST_PLATFORM is x64
../../ThirdParty/PSCommon/BuildSystem/CommonDefs.mak:40: HOST_PLATFORM is x64
../../ThirdParty/PSCommon/BuildSystem/CommonDefs.mak:40: HOST_PLATFORM is x64
../../../ThirdParty/PSCommon/BuildSystem/CommonDefs.mak:40: HOST_PLATFORM is x64
Formats/XnFormatsMirror.cpp: In function ‘XnStatus XnMirrorOneBytePixels(XnUChar*, XnUInt32, XnUInt32)’:
Formats/XnFormatsMirror.cpp:46:11: error: array subscript is below array bounds [-Werror=array-bounds]
  XnUInt8* pDestEnd = &pLineBuffer[0] - 1;
           ^~~~~~~~
Formats/XnFormatsMirror.cpp: In function ‘XnStatus XnMirrorTwoBytePixels(XnUChar*, XnUInt32, XnUInt32)’:
Formats/XnFormatsMirror.cpp:79:12: error: array subscript is below array bounds [-Werror=array-bounds]
  XnUInt16* pDestEnd = &pLineBuffer[0] - 1;
            ^~~~~~~~
Formats/XnFormatsMirror.cpp: In function ‘XnStatus XnMirrorThreeBytePixels(XnUChar*, XnUInt32, XnUInt32)’:
Formats/XnFormatsMirror.cpp:115:11: error: array subscript is below array bounds [-Werror=array-bounds]
  XnUInt8* pDestEnd = &pLineBuffer[0] - 1;
           ^~~~~~~~
cc1plus: all warnings being treated as errors
../../../ThirdParty/PSCommon/BuildSystem/CommonCppMakefile:141: recipe for target '../../../Bin/Intermediate/x64-Release/liborbbec.so/XnFormatsMirror.o' failed
make[4]: *** [../../../Bin/Intermediate/x64-Release/liborbbec.so/XnFormatsMirror.o] Error 1
Makefile:131: recipe for target 'Source/Drivers/orbbec' failed
make[3]: *** [Source/Drivers/orbbec] Error 2
ros_astra_camera/CMakeFiles/astra_openni2.dir/build.make:110: recipe for target 'ros_astra_camera/astra_openni2/src/astra_openni2-stamp/astra_openni2-build' failed
make[2]: *** [ros_astra_camera/astra_openni2/src/astra_openni2-stamp/astra_openni2-build] Error 2
CMakeFiles/Makefile2:1018: recipe for target 'ros_astra_camera/CMakeFiles/astra_openni2.dir/all' failed
make[1]: *** [ros_astra_camera/CMakeFiles/astra_openni2.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

I have ros_astra_cam and usb_cam with the launch file from the other astra git in my catkin_astra/src directory.
It's the same I have with my other Ubuntu system except that the other one is a Mint 18.3 with ROS Kinetic and now its Mint 19 with ROS Melodic

mzwang828, 2018-07-20:
Hi, 
Have you solved this problem? I'm having the same error with ROS Melodic

BreakingLaw, 2019-02-04:
anyone can solve this problem?, im working with ros melodic too.

Nan-Orbbec3d-US, 2019-06-09:
Hi guys, could you try the latest version again. I try to build it with melodic + ubuntu 18.04 and it works. If you still have the same error, please let me know.

NikolasE, 2019-07-31:
Works for me on Melodic

Nan-Orbbec3d-US, 2019-07-31:
@NikolasE Cool, I would like to close this. If anyone has problem on Melodic, feel free to reopen it.

closed 2019-07-31 by Nan-Orbbec3d-US

ANABUR920, 2019-08-20:
still cannot work on ubuntu 18 + melodic, and the same error

Nan-Orbbec3d-US, 2019-08-20:
Could you check the version of the package? According to the log, the build system tried to build XnLib and PSCommon which were removed after 0.3.0.

bonedoz, 2020-11-29:
@Nan-Orbbec3d-US Can't work on `ROS Melodic` & `GCC 7.5.0` with the latest version of the package.

KIYOLY, 2023-04-25:
this still happens to me
(jetson nano, melodic)