orbbec/ros_astra_camera #64: Changing the depth mode

kelvinquan, 2018-08-21:
Hi,

Have just purchased the Orbec Astra Mini and is testing it out using ros. The default depth_mode is 30hz (VGA). I would like to know in which part of the program should I change to reduce it to 25hz?

Cheers

chataign, 2018-09-14:
I believe that the Astra can only produce data for a fixed set of rates and resolutions (Orbbec support can confirm). That being said you can easily downsample your depth stream using topic_tools for instance

http://wiki.ros.org/topic_tools/throttle

Nan-Orbbec3d-US, 2019-06-28:
To modify the resolution, you can change the default video mode in `Astra.cfg` as below
```sh
gen.add("ir_mode", int_t, 0, "Video mode for IR camera", 5, 1, 14, edit_method = output_mode_enum)
gen.add("color_mode", int_t, 0, "Video mode for color camera", 5, 1, 14, edit_method = output_mode_enum)
gen.add("depth_mode", int_t, 0, "Video mode for depth camera", 5, 1, 14, edit_method = output_mode_enum)
```
Modify the default value from 5 to 6(25fps). Then after recompilation, you should be able to run cameras with 640X480@25fps. Or you can switch the setting in dynamic reconfiguration.

BUT, **Astra Mini doesn't support 25fps**. Therefore, I would recommend you postprocess the data by yourself.

closed 2019-06-28 by Nan-Orbbec3d-US