orbbec/ros_astra_camera #66: All rgb topics are empty

frushton, 2018-10-12:
When I launch the astrapro.launch file and do a 'rostopic list', it lists all of the depth, ir, and rgb topics. The depth and ir topics are being published to, and I can view the image feed. However, the rgb topics are all empty. This is the terminal I launch the launch file in:

roslaunch astrapro.launch 
... logging to /home/e4e/.ros/log/ace1494e-cdb3-11e8-af2b-9cb6d0dcfaeb/roslaunch-e4e-XPS-15-9560-17311.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://e4e-XPS-15-9560:40203/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/driver/auto_exposure: True
 * /camera/driver/auto_white_balance: True
 * /camera/driver/bootorder: 0
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: False
 * /camera/driver/device_id: #1
 * /camera/driver/devnums: 1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    driver (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)
  /
    camera_base_link (tf/static_transform_publisher)
    camera_base_link1 (tf/static_transform_publisher)
    camera_base_link2 (tf/static_transform_publisher)
    camera_base_link3 (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [17321]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ace1494e-cdb3-11e8-af2b-9cb6d0dcfaeb
process[rosout-1]: started with pid [17334]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [17345]
process[camera/driver-3]: started with pid [17352]
process[camera/rgb_debayer-4]: started with pid [17353]
process[camera/rgb_rectify_mono-5]: started with pid [17354]
process[camera/rgb_rectify_color-6]: started with pid [17364]
process[camera/depth_rectify_depth-7]: started with pid [17371]
process[camera/depth_metric_rect-8]: started with pid [17388]
process[camera/depth_metric-9]: started with pid [17393]
process[camera/depth_points-10]: started with pid [17403]
process[camera/register_depth_rgb-11]: started with pid [17407]
process[camera/points_xyzrgb_sw_registered-12]: started with pid [17418]
process[camera/depth_registered_sw_metric_rect-13]: started with pid [17422]
process[camera_base_link-14]: started with pid [17430]
process[camera_base_link1-15]: started with pid [17444]
[ INFO] [1539303222.469148798]: Initializing nodelet with 4 worker threads.
process[camera_base_link2-16]: started with pid [17461]
process[camera_base_link3-17]: started with pid [17477]
[ INFO] [1539303222.692063845]: Device "2bc5/0403@1/8" found.
Warning: USB events thread - failed to set priority. This might cause loss of data...

Is this being caused by the "failed to set priority" warning? Or is this something else? 

Also, I have run the creat_udev_rules script included.

vishu2287, 2018-10-16:
hello @frushton . i am encountering same problem. May i know if it has been solved?

frushton, 2018-10-18:
Hello @vishu2287, it seems that the problem comes from the fact that the rgb stream of the Astra Pro camera is only accessible through the UVC interface. This link discusses this problem, and offers some potential solutions that I have not tried yet. 

https://3dclub.orbbec3d.com/t/compatibility-astra-pro-and-openni2/108/2

Nan-Orbbec3d-US, 2019-06-26:
Hi @frushton , @vishu2287 , could you try the latest version again. Thank you!

Nan-Orbbec3d-US, 2019-08-01:
fixed after commit 67a44c44105eb4c86d3a8476a10f8c54ef1f0895

closed 2019-08-01 by Nan-Orbbec3d-US