orbbec/ros_astra_camera #72: embedded_s not detected

NikolasE, 2019-06-25:
I tried to test a new embedded S camera (label AQAAC8, 30009) with this driver. If I connect the camera to a USB3 port on my computer, I get this dmesg: 

usb 8-1: new SuperSpeed USB device number 4 using xhci_hcd
[113807.293831] usb 8-1: New USB device found, idVendor=05e3, idProduct=0620
[113807.293832] usb 8-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[113807.293833] usb 8-1: Product: USB3.1 Hub
[113807.293834] usb 8-1: Manufacturer: GenesysLogic
[113807.294522] hub 8-1:1.0: USB hub found
[113807.294837] hub 8-1:1.0: 2 ports detected
[113807.368571] usb 7-1.1: new high-speed USB device number 29 using xhci_hcd
[113807.471177] usb 7-1.1: New USB device found, idVendor=05e3, idProduct=0610
[113807.471179] usb 7-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[113807.471180] usb 7-1.1: Product: USB2.1 Hub
[113807.471180] usb 7-1.1: Manufacturer: GenesysLogic
[113807.471868] hub 7-1.1:1.0: USB hub found
[113807.472128] hub 7-1.1:1.0: 2 ports detected
[113807.581012] usb 8-1.2: new SuperSpeed USB device number 5 using xhci_hcd
[113807.601488] usb 8-1.2: New USB device found, idVendor=2bc5, idProduct=060b
[113807.601490] usb 8-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[113807.601491] usb 8-1.2: Product: ORBBEC Depth Sensor
[113807.601491] usb 8-1.2: Manufacturer: Orbbec(R)
[113807.784578] usb 7-1.1.1: new high-speed USB device number 30 using xhci_hcd
[113807.928807] usb 7-1.1.1: New USB device found, idVendor=2bc5, idProduct=050b
[113807.928809] usb 7-1.1.1: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[113807.928819] usb 7-1.1.1: Product: USB 2.0 Camera
[113807.928820] usb 7-1.1.1: Manufacturer: Sonix Technology Co., Ltd.
[113807.928820] usb 7-1.1.1: SerialNumber: SN0001
[113807.930632] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (2bc5:050b)
[113807.945564] uvcvideo 7-1.1.1:1.0: Entity type for entity Extension 4 was not initialized!
[113807.945567] uvcvideo 7-1.1.1:1.0: Entity type for entity Extension 3 was not initialized!
[113807.945568] uvcvideo 7-1.1.1:1.0: Entity type for entity Processing 2 was not initialized!
[113807.945570] uvcvideo 7-1.1.1:1.0: Entity type for entity Camera 1 was not initialized!
[113807.945750] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1c.3/0000:04:00.0/usb7/7-1/7-1.1/7-1.1.1/7-1.1.1:1.0/input/input25

The devices are created according to the udev-rules:

foo@barhostname:/dev$ ls
astrauvc         deeyea (...)

I can open the rgb-camera as a webcam. 

_astra_list_devices_ however does not find the device. 

What should I try next?

Nan-Orbbec3d-US, 2019-06-26:
Are you using the latest version of this package? And could you provide the log output by Orbbec driver? You can turn on these two lines in `include/openni2_redist/x64/OpenNI.ini`(if you are using x64 linux).
```
Verbosity=0
LogToConsole=1
```
Then run the *astra_list_devices* program again and record the log.

I think you could also download the [astra_launch](https://github.com/orbbec/ros_astra_launch) package and try `roslaunch astra_launch embedded_s.launch` to see if there are any depth topics appeared.

NikolasE, 2019-06-27:
Hey! I'm sorry, I linked against an older orbbec_openni-version that was still on my computer. The camera is now found.

closed 2019-06-27 by NikolasE