orbbec/ros_astra_camera #73: Stereo S - Only One IR Camera Stream Availabe

faizan-m, 2019-06-26:
At the moment there is only one IR camera stream available in the driver but Stereo S has two IR cameras, is it possible to enable that as well or will it be available in the future?

Nan-Orbbec3d-US, 2019-06-26:
This package is developed based on Orbbec OpenNI2 SDK and we cannot enable the other IR camera by SDK for now. Probably it will be available in the future.

Nan-Orbbec3d-US, 2019-08-01:
Hi @faizan-m , the SDK will support switching IR cameras (left/right). I can add a service to switch IR cameras. However, we only have the **switch**, it means we still cannot grab images simultaneously. Hope this helps.

faizan-m, 2019-08-01:
Yeah that would be helpful! I am assuming the service will also switch the tf of the optical frame as well?

Nan-Orbbec3d-US, 2019-08-01:
Good idea, but I do not know how to get the extrinsics between left and right IR. Thus, it won't switch the tf. However, I will keep it in my mind.

Nan-Orbbec3d-US, 2019-08-09:
Hi, @faizan-m , you can check service `/camera/switch_ir_camera` in the latest version now ;-)

Nan-Orbbec3d-US, 2019-08-19:
Hi, @faizan-m , did you try this new function? If you can get both IR images, I think we can close this issue.

faizan-m, 2019-08-20:
For my particular use case, I was interested in using both the image streams simultaneously to do stereoscopic stuff. While the service opens up the other stream as well, ideally I would want both at the same time if possible

Nan-Orbbec3d-US, 2019-08-21:
Ok, I will keep your requirement in mind. Once SDK supports it, I will add this feature.