orbbec/ros_astra_camera #76: Factory calibration not loaded for MINI_S

NikolasE, 2019-06-28:
According to this answer in the forum, the MINI_S should also be factory calibrated, but it is not read by the driver:

https://3dclub.orbbec3d.com/t/is-the-embedded-s-factory-calibrated/2535

Mini S:

    D: [0.0, 0.0, 0.0, 0.0, 0.0]
    K: [570.3400268554688, 0.0, 319.5, 0.0, 570.3400268554688, 239.5, 0.0, 0.0, 1.0]
    R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
    P: [570.3400268554688, 0.0, 319.5, 0.0, 0.0, 570.3400268554688, 239.5, 0.0, 0.0, 0.0, 1.0, 0.0]

Embedded:

    D: [0.038348376750946045, -0.059544675052165985, -0.0007290133507922292, -0.0005021999240852892, 0.0]
    K: [-455.1952209472656, 0.0, 327.48583984375, 0.0, 455.1952209472656, 240.41822814941406, 0.0, 0.0, 1.0]
    R: [0.9999920129776001, -0.0036066046450287104, -0.0017094686627388, 0.003609547158703208, 0.9999920129776001, 0.0017212817911058664, 0.0017032469622790813, -0.0017274385318160057, 0.999997079372406]
    P: [455.1952209472656, 0.0, 312.51416015625, -14.494669914245605, 0.0, 455.1952209472656, 240.41822814941406, 0.03783918172121048, 0.0, 0.0, 1.0, -0.5576953887939453]

Sayter99, 2019-06-30:
Hi @NikolasE ,

Currently, we use different methods to fetch intrinsic parameters. For the latest cameras (stereo_s, embedded_s), we adopt Orbbec3d extended APIs to get parameters. However, we didn't use this method for previous cameras. In the future, we will try to find a way to use the same APIs to get parameters. But, for now, I recommend you to use Extend API by yourself to get MINI_S' intrinsic parameters.

NikolasE, 2019-07-01:
Could you give me some more details? Is there a current documentation of the SDK? The docs in the SDK download is rather short. Reading the intrinsic parameters is a fundamental function of the SDK. Is there an easier way to get the intrinsics (e.g. on Windows) so that we could  store it on the file system (which is not elegant, but should work)

Nan-Orbbec3d-US, 2019-07-01:
After launching your astra camera, you can use `rosservice call /camera/get_camera_info "{}"` to get intrinsic parameters. You can also trace this service for understanding how to use Extended API. BTW, in windows, there is a sample code of Extended API in Orbbec OpenNI SDK without intrinsic parameters reading, but you could add that by following the Extended API documentation in `Extended API folder`.

closed 2019-07-02 by Nan-Orbbec3d-US

NikolasE, 2019-07-03:
Why is the Astra handled differently than the other cameras? 

in AstraDriver::getColorCameraInfo:

    if (device_->getDeviceTypeNo() == OB_STEREO_S_NO || device_->getDeviceTypeNo() == OB_EMBEDDED_S_NO ||
        device_->getDeviceTypeNo() == OB_ASTRA_PRO_NO ||
        /// load factory calibration

Nan-Orbbec3d-US, 2019-07-03:
because gen1 cameras (astra, astra pro, mini) are different from gen2 (stereo s, embedded s). gen2 cameras' architecture is a depth sensor + a uvc. so, we need to use another api to get colorCameraInfo of that uvc. if you still feel unclear, feel free to reopen it :)

vigneshrajap, 2020-12-29:
Even Astra Stereo U3 is not loading factory calibration. 

Can anybody look into it?