orbbec/ros_astra_camera #77: High Resolution Depth Image on Embedded

NikolasE, 2019-07-01:
I tried to extract some high resolution depth images from my embedded and found some issues:

- Astra.cfg has to entry for 1280/800 Resolution
- void AstraDriver::configCb ignores depth_mode and uses mode 13
- Official data sheet:
> 1280×800@30FPS
> 640×400@60FPS
But the camera offers only:
> Depth sensor video modes:
>    - Resolution: 640x480@30Hz Format: Depth 1mm
>    - Resolution: 640x480@30Hz Format: Depth 100um
>    - Resolution: 640x400@30Hz Format: Depth 1mm
>    - Resolution: 640x400@30Hz Format: Depth 100um
>    - Resolution: 1280x800@7Hz Format: Depth 1mm
>    - Resolution: 1280x800@7Hz Format: Depth 100um

If I add a new video mode:
```

  // 1280×800_30Hz
  video_mode.x_resolution_ = 1280;
  video_mode.y_resolution_ = 800;
  video_mode.frame_rate_ = 7;
  video_modes_lookup_[15] = video_mode;
```

The camera accepts this depth_mode and publishes a corresponding camera_info, however, the data is scrambled:
[image]

This is my arm in front of a wall and the strips are scrolling through the window.

Nan-Orbbec3d-US, 2019-07-02:
If you don't need color point cloud, you can simply turn off the depth_registration in `embedded_s.launch`.
`<arg name="depth_registration" default="false" />`
Then, it should be fine. If you still need color point cloud, you need to set uvc with higher resolution (close to 1280x800).

NikolasE, 2019-07-03:
How is this behaviour related to depth_registration? Do you want to say that it is not possible to use the higher resolution with depth_registration?

Nan-Orbbec3d-US, 2019-07-03:
It's all about the `registration support` from the driver. You can check the file `ros_astra_camera/include/openni2_redist/x64/OpenNI2/Drivers/orbbec.ini`. There is an option `RegistrationType=0`. It usually uses `hardware acceleration mode`. However, the hardware doesn't support accelerating 1280x800 image registration. You can firstly try to switch it to `software registration` and take a look at the result. If you don't satisfy with the performance of the built-in software registration. You need to do it by yourself, which is the way I recommended. You need to get the extrinsic parameters and check PCL or pcl_ros tutorials to figure out how to create a point cloud. So, I would like to say that it is possible to use the higher resolution with depth_registration. But you need to have extremely high computing horsepower (because you won't have hardware acceleration).

closed 2019-07-11 by Nan-Orbbec3d-US