orbbec/ros_astra_camera #78: No ColorSensor found on Embedded S

NikolasE, 2019-07-01:
I just noticed that my Embedded S lost its color sensor...

If I add an cout << *device_ << endl; I get this:

> Device info (2bc5/060b@7/8)
>    Vendor: Orbbec
>    Name: Astra
>    USB Vendor ID: 11205
>    USB Product ID: 11205
> 
> IR sensor video modes:
>    - Resolution: 640x480@30Hz Format: Gray16
>    - Resolution: 640x480@30Hz Format: RGB888
>    - Resolution: 640x400@30Hz Format: Gray16
>    - Resolution: 640x400@30Hz Format: RGB888
>    - Resolution: 1280x800@7Hz Format: Gray16
>    - Resolution: 1280x800@7Hz Format: RGB888
> No Color sensor available
> Depth sensor video modes:
>    - Resolution: 640x480@30Hz Format: Depth 1mm
>    - Resolution: 640x480@30Hz Format: Depth 100um
>    - Resolution: 640x400@30Hz Format: Depth 1mm
>    - Resolution: 640x400@30Hz Format: Depth 100um
>    - Resolution: 1280x800@7Hz Format: Depth 1mm
>    - Resolution: 1280x800@7Hz Format: Depth 100um

The uvc_camera-node however is able to get data from the camera:

> $rosrun uvc_camera uvc_camera_node 
> [/uvc_camera  INFO 1561997861.385158453]: using default calibration URL
> [/uvc_camera  INFO 1561997861.385955174]: camera calibration URL: file:///home/asd/.ros/camera_info/camera.yaml
> [/uvc_camera  INFO 1561997861.386040240]: Unable to open camera calibration file [/home/asd/.ros/camera_info/camera.yaml]
> [/uvc_camera  WARN 1561997861.386075859]: Camera calibration file /home/asd/.ros/camera_info/camera.yaml not found.
> opening /dev/video0
> pixfmt 0 = 'MJPG' desc = 'Motion-JPEG'
>   discrete: 2592x1944:   1/25 
>   discrete: 2560x1440:   1/25 
>   discrete: 1920x1080:   1/30 
>   discrete: 1280x960:   1/30 
>   discrete: 1280x720:   1/30 
>   discrete: 640x480:   1/30 
>   discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
> pixfmt 1 = 'YUYV' desc = 'YUYV 4:2:2'
>   discrete: 2592x1944:   1/2 
>   discrete: 2560x1440:   1/2 
>   discrete: 1920x1080:   1/5 
>   discrete: 1280x960:   1/5 
>   discrete: 1280x720:   1/5 
>   discrete: 640x480:   1/25 
>   discrete: 320x240:   1/30 
>   int (Brightness, 0, id = 980900): -64 to 64 (1)
>   int (Contrast, 0, id = 980901): 0 to 64 (1)
>   int (Saturation, 0, id = 980902): 0 to 128 (1)
>   int (Hue, 0, id = 980903): -40 to 40 (1)
>   bool (White Balance Temperature, Auto, 0, id = 98090c): 0 to 1 (1)
>   int (Gamma, 0, id = 980910): 72 to 500 (1)
>   int (Gain, 0, id = 980913): 0 to 100 (1)
>   menu (Power Line Frequency, 0, id = 980918): 0 to 2 (1)
>     0: Disabled
>     1: 50 Hz
>     2: 60 Hz
>   int (White Balance Temperature, 16, id = 98091a): 2800 to 6500 (1)
>   int (Sharpness, 0, id = 98091b): 0 to 6 (1)
>   int (Backlight Compensation, 0, id = 98091c): 0 to 65535 (1)
>   menu (Exposure, Auto, 0, id = 9a0901): 0 to 3 (1)
>   int (Exposure (Absolute), 16, id = 9a0902): 1 to 5000 (1)
>   bool (Exposure, Auto Priority, 0, id = 9a0903): 0 to 1 (1)

The camera_node from the ros_astra_camera package also has problems:
rosrun astra_camera camera_node 

> [/libuvc_camera  INFO 1561997977.632824247]: Opening camera with vendor=0x0, product=0x0, serial="", index=0
> unsupported descriptor subtype VS_COLORFORMAT
> uvc_get_stream_ctrl_format_size: Invalid mode (-51)
> [/libuvc_camera ERROR 1561997977.710191163]: check video_mode/width/height/frame_rate are available
> Segmentation fault

Could this somehow be related to my last tests with the high resolution depth images?

Nan-Orbbec3d-US, 2019-07-02:
We will integrate an uvc framework in Orbbec SDK in the future, but, for now, you cannot see color sensors in OpenNI devices.

BTW, You need to specific video_mode/width/height/frame_rate while using `astra_camera camera_node`.

NikolasE, 2019-07-03:
This was closed a bit too fast. You mean we are not able to stream color images from the driver node?

Nan-Orbbec3d-US, 2019-07-03:
We cannot stream color images by OpenNI SDK now, so we use libuvc to grab color images.

closed 2019-07-08 by Nan-Orbbec3d-US