orbbec/ros_astra_camera #80: Stereo S - Toggling depth registration changes apparent focal length

coxep, 2019-07-09:
I'm running the Stereo S with the latest code. I noticed that if I toggle the  /camera/driver/depth_registration parameter in dynamic reconfigure, the location of projected features in 3D change. I suspect this is due to the misconfigured intrinsic parameters for one of these modes. The published intrinsics do change as this setting is toggled as expected (the image changes size and appears cropped around the edges) Which intrinsics are correct? They both can't be, since unique features in the 3d cloud project to noticeably different locations.

Nan-Orbbec3d-US, 2019-07-10:
Hi @coxep , if you disable depth_registration, the driver will grab default parameters from OpenNI API. However, the default values from OpenNI is rough. If you do need accurate parameters, you should enable depth_registration.

coxep, 2019-07-16:
Hi @Nan-Orbbec3d-US. Looking at the code, it appears that if depth_registration is enabled, the driver grabs the RGB parameters (see line 641 in astra_driver.cpp). I don't understand the rationale for this. 

From your comment above, it looks like the only options for depth intrinsics are either the default intrinsics (depth_registration=false), or intrinsics for the RGB camera (depth_registration=true). Am I understanding this correctly?

Nan-Orbbec3d-US, 2019-07-17:
Yes, you are right. To do depth registration, we need to align rgb and depth frame_ids. So, as you saw, the behavior is different while doing depth registration.

coxep, 2019-07-18:
This does not make sense to me. I think the correct way to do depth registration would be to map the RGB image to the depth image by changing the RGB intrinsics, not the other way around. Using RGB intrinsics for the depthimage is guaranteed to make the 3D projection of the depthimage wrong, since the correct depth intrinsics are derived by the IR cameras.

Nan-Orbbec3d-US, 2019-07-18:
Agreed. This package is designed for gen 1 cameras and there are some weird behavior we should check. Probably, we need to refactor it thoroughly. Anyways, I will check this. Thank you for reporting this.

coxep, 2019-07-23:
So right now, there is no way to access the correct intrinsics (which the stereo S uses internally) either for the depth image or for the IR image?

Nan-Orbbec3d-US, 2019-07-23:
For the 640x400 resolution, all images are undistorted by the firmware. To get the intrinsics of this resolution, you can use Orbbec Extended API. It's contained in [Orbbec OpenNI SDK](http://dl.orbbec3d.com/dist/openni2/OpenNI_2.3.0.55.zip). You can find the `Extended API` directory and a document in that directory. You can follow the documentation to get the intrinsics for IR image (left one).

coxep, 2019-07-26:
The command listed in the Extended SDK documentation (Page 2, section 6: Obtain Camera Parameters) uses the exact same command currently used in the astra_driver for obtaining the RGB camera params. I am interested in getting RGB, IR, and depth intrinsics, not just RGB. How do I get intrinsic parameters for each of these streams?

coxep, 2019-07-29:
Thank you!

Left corresponds to the IR image. Does right correspond to the RGB image? (It definitely seems to from looking at astra_driver.cpp.)

Finally, Is there a way to extract the depthimage intrinsics?

Nan-Orbbec3d-US, 2019-07-29:
Yes, left corresponds to the IR and right corresponds to the RGB. The depth data is derived from IR data, so the depth image doesn't have it own intrinsics.

coxep, 2019-07-31:
I'm confused by "the depthcam doesn't have its own intrinsics". We have established above that if the intrinsics for the depthimage are incorrect, then the projection to the depthcloud will be incorrect. 

Let me rephrase my question: How do I access the projection matrix that I will need to **accurately** project the depthimage into a 3D pointcloud? I understand that this will be a function of the IR cameras' intrinsics, but unless the depthimage published by astra_driver has already been reprojected to the left IR image, I need more than just the left IR camera's intrinsics for this.

Nan-Orbbec3d-US, 2019-07-31:
The other IR is aligned to the IR camera by default. So, by using the intrinsics from Extended API, you should be able to create an accurate point cloud.

Let me know more about your goal. Are you trying to increase the resolution and then generate a denser point cloud? If so, you should use two IR images, calibrate them (both intrinsics and extrinsics), and then write an algorithm to create point cloud.

coxep, 2019-07-31:
Are you saying that the left IR image should perfectly align with depthimage?

I am working to update our automatic extrinsic calibration process for our robots to work with the stereo S cameras. My end goal is ensure that the 3D pointcloud generated from the stereo S camera aligns with reality as much as possible (for accurate obstacle pose identification, etc)

Here is the process, which I think is pretty standard for extrinsic calibration:

- Place the robot at a known pose relative to a calibration board.
- In the IR image, identify known features on the calibration board.
- Given the mapping from IR to depth image (using intrinsics and relative shift), project known features from IR image to depth image
- With corresponded features from depth image to calibration board, solve for camera extrinsics (camera to board) using homography.
- Using known pose from board to robot, solve for robot to depth optical transform.

To do this effectively, I need to be able to map the pixel location in the IR camera to a corresponding pixel location in the depth image.

I am also working on an identical approach, using the RGB camera instead of the IR. So, I am also very interested in the mapping from RGB to depth image. Historically I've used IR, because many cameras do not have RGB images, or the intrinsics have not been determined. And, it is often much more trivial to map from IR to depth, since depth is derived from IR.

Thank you for your responsiveness, I greatly appreciate it :)

Nan-Orbbec3d-US, 2019-07-31:
Yes, you should be able to generate a good point cloud by depth and left ir intrinsics.

Cool! Your work is decent. But I am curious that how could you get a clean IR image? Aren't there a lot of spots on your IR image? If you cannot get a clean IR image, it should be hard to do feature matching. If you feel okay, could you share your method about dealing with this problem?

To map your result to IR camera, you can try this. I don't verify it by myself before, but it works theoretically. Remember that we can get extrinsics `r2l` from extended API. It contains rotation matrix and translation from uvc to left ir. So, you can simply transform your results (in extrinsic calibration, you will get a rotation matrix and translation vector between two cameras) by these value. For example, you can assume your camera1's position is (0, 0, 0) and orientation is (0, 0, 0) (axis angle) as well. Then, you can get your camera2's position by your calibration results. Next, transform these two **uvc** pose to **ir** pose. Finally, solve the transform between them. However, you should be aware of that this process might enlarge the error because of errors from our extrinsics (uvc to ir).

coxep, 2019-07-31:
Thanks! Yes, that's exactly how I was planning to do this :) It all hinged on the depthimage sharing the left IR intrinsics, and it seems like it does. I captured some frames from IR and depth, and the overlay looks near-perfect :)

To get a clean IR image, I am using a light source that hangs above the board and illuminates it. I also disable the IR emitters on the depthcams. Here is an IR image that was captured during extrinsic calibration for the Astra mini, using an 850nm illuminator:

[image]

For feature correspondence, I am creating a graph to track each corner (I know the precise location of the center). This is very similar to the opencv library, except I don't need a fixed grid size.

Nan-Orbbec3d-US, 2019-07-31:
It's cool! Such a big calibration board. I am developing an extrinsic calibration algorithm, too. At first, I want to use opencv chessboard as well, but it force me to know the board size. To apply it to our non-overlapping system is impossible. It's good to know that you are developing chessboard detection algorithm. Will it be opened?

I currently use a board with circles designed by myself and solve the essential matrix to get the R, t. BTW, you can try ceres-solver to optimize your calibration results, it's extremely useful. If you feel fine, we can talk more about this via email (cjwu@orrbec3d.com). (because this is not relevant to this issue haha)

closed 2019-08-01 by Nan-Orbbec3d-US