orbbec/ros_astra_camera #83: Camera Node Crashes

faizan-m, 2019-07-10:
The camera node on my system crashes randomly with the following error:

`[camera/stereo_s-6] process has died [pid 9094, exit code -11, cmd /home/faizan/catkin_ws/devel/lib/astra_camera/camera_node /camera/image_raw:=/camera/rgb/image_raw __name:=stereo_s __log:=/home/faizan/.ros/log/3551285c-a325-11e9-ab44-309c23644e94/camera-stereo_s-6.log].`

The log file is empty and there are no error messages printed before this crash. Any ideas how to approach debugging this?

Nan-Orbbec3d-US, 2019-07-10:
Hi @faizan-m , are you using multiple cameras simultaneously? Or could you provide more details such as your launch file? It usually happens while you are using a wrong uvc setting (format, frame rate, resolution). The recommended setting for stereo_s is `640x480@30fps with yuyv`.

faizan-m, 2019-07-10:
Hi!
Thanks for getting back so quickly! I am only using a single stereo s camera. My launch file is derived from the original stereo-s launch file:

https://gist.github.com/faizan-m/a186a58379866b6b116bd0a20b3939cd

faizan-m, 2019-07-10:
I think I found it in line 74:
`<param name="camera_info_url" value=""/>`
I added the camera calibration url here as well (apart from setting it in rgb_camera_info_url parameter:

And that seems to have fixed it.

faizan-m, 2019-07-10:
It actually still crashes. Much less frequently than before though....

Nan-Orbbec3d-US, 2019-07-10:
I've already checked your file. It should be fine. For now, I would recommend you to change another usb cable. According to [this issue](https://3dclub.orbbec3d.com/t/astra-stereo-s-with-ros-unable-to-access-depth-data/2513/4), I think low-quality cables are possible to produce unexpected crashes.

Nan-Orbbec3d-US, 2019-07-18:
Hi @faizan-m , do you fix the problem? If not, you could modify the launch file by switching `yuyv` to `mjpeg` and try it again. Hope it helps.

faizan-m, 2019-07-20:
The problem slowly disappeared so I am assuming it is because of the USB cable but I will keep that in mind, thanks!

closed 2019-07-20 by faizan-m