orbbec/ros_astra_camera #84: Camera intrinsics do not change with resolution

ruffner, 2019-07-11:
The camera (Stereo S) intrinsics matrix that are a part of the camera info header are not being updated by the driver whenever different resolutions are selected (after reconfigure and relaunch).

This bug appears under ROS Kinetic running the most recent version of the Orbbec driver as of today (commit hash 7832c4242b987a6a3d4bc69b95beea6cd3e6f1d6).

I believe this is related to issue #50, however it is much more general and creates a problem for anyone trying to use the Stereo S at any resolution besides 640x400.

For example. When using a resolution of 640x400, the intrinsics of the header look like
```
---
header:
  seq: 81
  stamp:
    secs: 1562872724
    nsecs: 205047600
  frame_id: "depthcam_proximity_front2_depth_optical_frame"
height: 400
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [540.9197387695312, 0.0, 318.4385070800781, 0.0, 477.4271545410156, 184.38267517089844, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [540.9197387695312, 0.0, 318.4385070800781, 0.0, 0.0, 477.4271545410156, 184.38267517089844, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 1
binning_y: 1
---
```

After reconfiguring and relaunching, the same `rostopic echo` command shows that the resolutions have changed, but the camera intrinsics have not (except for the 6th element of K):
```
--
header:
  seq: 106
  stamp:
    secs: 1562872426
    nsecs:  19165985
  frame_id: "depthcam_proximity_front2_depth_optical_frame"
height: 200
width: 320
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [540.9197387695312, 0.0, 318.9385070800781, 0.0, 477.4271545410156, 189.38267517089844, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [540.9197387695312, 0.0, 318.9385070800781, 0.0, 0.0, 477.4271545410156, 189.38267517089844, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 1
binning_y: 1
---
```

This is most easily seen in RViz, where after switching to the 320x200 resolution, the incorrect intrinsics cause the depth image to be rendered in an incorrect location in space. Using 640x400 resolution, the depth information is rendered correctly. Two images are shown below, the first is the correct depth registration when used at 640x400, the second is the same environmnent (with recognizable features), however, it is clear that the intrinsics are causing RViz to incorrectly render the depth data.

[image]

[image]

Nan-Orbbec3d-US, 2019-07-11:
Currently, we cannot get intrinsics with another resolution from Orbbec OpenNI SDK. So you should calibrate it by yourself.

ruffner, 2019-07-12:
Thank you for the prompt reply. What is the limiting factor here, are you pulling a default calibration matrix from OpenNI? Toying with `astra_driver.cpp` I discovered that a simple scaling and shifting of the focal distances and center point in the K matrix are enough to yield a fairly accurate rendering, in RViz, at least. Is this not something that can easily be built in.

[image]

For now, I can use the functionality of loading a camera calibration config from YAML to take care of this, which is great, but is slightly inconvenient if this is the intended functionality. If it is the intended functionality, then it should be more clearly documented.

Nan-Orbbec3d-US, 2019-07-17:
Tuning the original values probably works, but it's better to calibrate for different resolutions. However, I think that providing a nicer default value is a good idea, I'll do more research and think about this. Thank you. Lastly, the yaml is the the intended functionality. Will refine the documentation in the next version.

closed 2019-07-17 by ruffner