orbbec/ros_astra_camera #87: rviz sometimes can't display pointcloud2

Thuan98, 2019-07-27:
when i run : roscore, rosrun astra_camera astra_camera_node , roslaunch astra_launch astrapro.launch

Nan-Orbbec3d-US, 2019-07-27:
Hi @Thuan98 , could you provide more detailed information like the log? BTW, if you are working on a single computer, you can just use one command `roslaunch astra_camera astrapro.launch` to launch your astrapro.

Thuan98, 2019-07-29:
i use it on jetson tx2, but i can't visualize pointcloud2 on rviz. Status: received message,contains invalid floating point values (nans or infs) on camera.pointcloud2 also received message but not display

Nan-Orbbec3d-US, 2019-07-30:
Did you receive any specific error messages? So your procedure is
1. launch astrapro
2. echo topic `/camera/depth_registered/points`
3. get nans/infs
Is this correct?

iamzhuang, 2019-08-12:
I used the astro pro to run in ros...I could dispaly rgb and depth..but there was not any pointcloud to display in rviz..the rviz display the pointcloud2 topic messages is received,but not display in it.

spark@spark:~/astra_pro_ws/src/ros_astra_camera/launch$ roslaunch astra_camera astrapro.launch 
... logging to /home/spark/.ros/log/8eff2306-bcb4-11e9-8f3f-e4029b917850/roslaunch-spark-18939.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://spark:32993/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/camera_rgb/camera_info_url: 
 * /camera/camera_rgb/frame_rate: 30
 * /camera/camera_rgb/height: 480
 * /camera/camera_rgb/index: 0
 * /camera/camera_rgb/product: 0x0501
 * /camera/camera_rgb/serial: 0
 * /camera/camera_rgb/timestamp_method: start
 * /camera/camera_rgb/vendor: 0x2bc5
 * /camera/camera_rgb/video_mode: yuyv
 * /camera/camera_rgb/width: 640
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /camera/driver/auto_exposure: True
 * /camera/driver/auto_white_balance: True
 * /camera/driver/bootorder: 0
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: True
 * /camera/driver/device_id: #1
 * /camera/driver/devnums: 1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    camera_rgb (astra_camera/camera_node)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    driver (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
  /
    camera_base_link (tf/static_transform_publisher)
    camera_base_link1 (tf/static_transform_publisher)
    camera_base_link2 (tf/static_transform_publisher)
    camera_base_link3 (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [18949]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8eff2306-bcb4-11e9-8f3f-e4029b917850
process[rosout-1]: started with pid [18962]
started core service [/rosout]
process[camera/camera_rgb-2]: started with pid [18979]
process[camera/camera_nodelet_manager-3]: started with pid [18980]
[... 255 lines cut ...]
 14624300 VERBOSE    Firmware.FrameSync (1) was successfully set.
 14624321 VERBOSE    Firmware.Stream1Mode (6) was successfully set.
 14624327 VERBOSE    Setting Firmware.CloseRange (84) to 0...
 14624688 VERBOSE    Firmware.CloseRange (84) was successfully set.
 14624697 VERBOSE    Setting Firmware.DepthCropEnabled (55) to 0...
 14625021 VERBOSE    Firmware.DepthCropEnabled (55) was successfully set.
 14625029 INFO       Property Depth.State was changed to 1.
 14688102 WARNING    Depth: Expected 1, got b43f
 14688198 WARNING    Depth frame is corrupt!
 14727364 VERBOSE    [FPS] IR: 0.00 Depth: 0.00 
[ INFO] [1565581945.422387886]: using default calibration URL
[ INFO] [1565581945.422482302]: camera calibration URL: file:///home/spark/.ros/camera_info/depth_Astra_Orbbec.yaml
[ INFO] [1565581945.422629219]: Unable to open camera calibration file [/home/spark/.ros/camera_info/depth_Astra_Orbbec.yaml]
[ WARN] [1565581945.422706374]: Camera calibration file /home/spark/.ros/camera_info/depth_Astra_Orbbec.yaml not found.
 15027559 VERBOSE    [FPS] IR: 0.00 Depth: 33.31 
 16003312 VERBOSE    [FPS] IR: 0.00 Depth: 30.56 
 17011455 VERBOSE    [FPS] IR: 0.00 Depth: 30.21 
 18027188 VERBOSE    [FPS] IR: 0.00 Depth: 30.00 
 19003621 VERBOSE    [FPS] IR: 0.00 Depth: 30.01 
 20011290 VERBOSE    [FPS] IR: 0.00 Depth: 30.00 
 21026374 VERBOSE    [FPS] IR: 0.00 Depth: 30.01 
 22002047 VERBOSE    [FPS] IR: 0.00 Depth: 30.02 
 23009457 VERBOSE    [FPS] IR: 0.00 Depth: 30.02 
 24021387 VERBOSE    [FPS] IR: 0.00 Depth: 30.05 
 25003145 VERBOSE    [FPS] IR: 0.00 Depth: 29.98

Nan-Orbbec3d-US, 2019-08-12:
Hi @iamzhuang , could you also provide the launch file you used? Thank you.

iamzhuang, 2019-08-13:
I pulled the newest src. and run the "astrapro.launch" of the project .    roslaunch astra_camera astrapro.launch

Nan-Orbbec3d-US, 2019-08-13:
Ok. Did you plug only one Astra Pro? According to your log, everything works fine. So, let's check
1. both the depth and color images are published and received correctly
2. did you choose the correct `fixed frame`, e.g. `camera_link`. make sure that you are not using `map` or `odom`.

Nan-Orbbec3d-US, 2019-08-19:
Hi @Thuan98 and @iamzhuang , did you resolve the problem?

iamzhuang, 2019-08-20:
nothing is resolved yet!    it could display rgb and depth image!!! so ,fixed frame, e.g. camera_link were chose right !   is the camera firmware version too old to support the pointcloud or others?

Nan-Orbbec3d-US, 2019-08-20:
could you get images from `/camera/rgb/image_rect_color` and `/camera/depth_registered/hw_registered/image_rect_raw`?

bool2007, 2019-08-29:
> could you get images from `/camera/rgb/image_rect_color` and `/camera/depth_registered/hw_registered/image_rect_raw`?
I have the same problem ,rostopic echo  nothing form the topic /camera/rgb/image_rect_color and /camera/depth_registered/hw_registered/image_rect_raw /camera/depth_registered/points

bool2007, 2019-08-29:
1180 WARNING    LibraryHandler: Couldn't find function oniDriverStreamConvertC2DCoordinates in libPSLink.so. Stopping
     1192 WARNING    Couldn't use file 'libPSLink.so' as a device driver

I found this warning after I lauch 
@Nan-Orbbec3d-US

liamincali, 2019-08-30:
Hello!

I would like to report a similar issue regarding the astrapro launch and rviz pointcloud visualization.

Rviz is receiving pointcloud information correctly, but does not display them. 

image_view returns the correct output for all topics camera/depth/... , as well as camera/depth_registered/image_raw, but does NOT display /camera/depth_registered/sw_registered/image_rect_raw. 

Trying to view raw image directly in Rviz results in Camera error: Contains invalid floating point values (nans or infs)

Below attached is the error I am referencing in rviz, as well as an rqt view of active nodes.
[image]
[image]

Thanks!

Sayter99, 2019-08-31:
@liamincali thank you so much! According to your log, I come up with a thing. Does your Astra Pro have intrinsics and extrinsics? You can echo `/camera/rgb/camera_info` and `/camera/depth/camera_info` to get relevant information.

Sayter99, 2019-08-31:
I don't have Astra Pro now, so I hope you guys can help me with checking the content of your `/camera/*/camera_info`. Now, I infer some of values are nan (not a number) so that the program cannot generate the point cloud. Thus, I fixed this in the latest version. I hope it would fix your problems. If the issue still exists, please let me know and give me more information. Thank you guys.

liamincali, 2019-08-31:
@Sayter99 I do not have access to the astra pro camera currently either, however I believe that the error is the NaN error that you described as well. I will report back as soon as I am able to test the new version, as well as camera_info if the problem still persists.

Thanks!

Sayter99, 2019-09-11:
@Thuan98 , @iamzhuang , @liamincali , @bool2007 . Hi guys, could anyone try the fix? Thank you for your time.

bool2007, 2019-09-20:
> @Thuan98 , @iamzhuang , @liamincali , @bool2007 . Hi guys, could anyone try the fix? Thank you for your time.
@Sayter99  I have try the new version ,with it rviz can display pointcloud
thank you very much!

liamincali, 2019-09-20:
@Sayter99 my bad, the point cloud is now being displayed. Thanks for the fix!

closed 2019-09-21 by Nan-Orbbec3d-US