orbbec/ros_astra_camera #88: no width/height set in get_camera_info service

NikolasE, 2019-07-31:
When calling the get_camera_info service, the width and height entry is missing (set to zero). Internally it uses convertAstraCameraInfo which looks like this:

```
sensor_msgs::CameraInfo AstraDriver::convertAstraCameraInfo(OBCameraParams p, ros::Time time) const
{
  sensor_msgs::CameraInfo info;
  // info.width = width;
  // info.height = height;
  info.distortion_model = sensor_msgs::distortion_models::PLUMB_BOB;
  info.D.resize(5, 0.0);
```

Nan-Orbbec3d-US, 2019-07-31:
This service is used to get UVC intrinsics and extrinsics (uvc to ir). Because we didn't integrate uvc and depth sensor yet, I cannot get uvc image size directly. The driver can be seen as two separate parts now. One is UVC (use `libuvc`), the other one is `astra_driver`. They can exchange information (yes, uvc can tell its image size to astra_driver then astra_driver can then publish it) via topics and services, however, it will cost more computing resources and we are still thinking how to integrate them nicely. To conclude, I won't let it publish image size until we finish the integration, because I cannot get them from uvc side elegantly. (or I need to add some weird exchanging mechanism in my perspective).

For now, I think it would be better to change a name of this service (because it's different from other `camera_info`). Do you have any recommendation? Thank you.

NikolasE, 2019-07-31:
I'm not quite sure if I understood you correcly. Is there another camera_info between rgb (uvc) and the depth image?

Nan-Orbbec3d-US, 2019-07-31:
You can get uvc intrinsics and extrinsics from `/camera/rgb/camera_info` topic and ir intrinsics from `/camera/ir/camera_info` topic.

Nan-Orbbec3d-US, 2019-08-19:
Hi @NikolasE , could you get the correct information from those topics?

brinij, 2020-04-02:
Do you know when will you integrate width and height of camera in get_camera_info service? Currently intrinsics are stored on the device but there is no other way to get them then through this service. 
`/camera/rgb/camera_info` topics publishes empty D if no `.ros/camera_info/rgb_Astra_Orbbec.yaml` is found, which means that camera needs to be calibrated intrinsically even though there is factory calibration on it. Is this correct?

closed 2020-12-08 by NikolasE