orbbec/ros_astra_camera #89: camera_link child frames are not static

faizan-m, 2019-08-12:
Hi!

Based on results from rqt_tf_tree, it seems that child frames of `camera_link` such as `camera_rgb_frame` and `camera_depth_frame` are not based on static transforms but rather are being published at 10Hz which is causing some problems in my system.

Is it correct that they are not static transforms?

faizan-m, 2019-08-12:
I found the code in  `astra_frames.launch` and changed the period from 100 to 1. That works better in my use case. 

I would recommend publishing the tf frame at atleast 30 Hz to match the camera FPS.

faizan-m, 2019-08-12:
Actually I found that using the tf2 static publisher works much better and at a much higher rate.

I would recommend updating the `astra_frames.launch` nodes to tf2 as well:

```  <node pkg="tf2_ros" type="static_transform_publisher" name="$(arg camera)_base_link"
    args="0 -0.02  0 0 0 0 $(arg tf_prefix)/$(arg camera)_link $(arg tf_prefix)/$(arg camera)_depth_frame" />  

  <node pkg="tf2_ros" type="static_transform_publisher" name="$(arg camera)_base_link1"
    args="0 -0.045 0 0 0 0 $(arg tf_prefix)/$(arg camera)_link $(arg tf_prefix)/$(arg camera)_rgb_frame" />  

  <node pkg="tf2_ros" type="static_transform_publisher" name="$(arg camera)_base_link2"
    args="$(arg optical_rotate) $(arg tf_prefix)/$(arg camera)_depth_frame $(arg tf_prefix)/$(arg camera)_depth_optical_frame" />  

  <node pkg="tf2_ros" type="static_transform_publisher" name="$(arg camera)_base_link3"
    args="$(arg optical_rotate) $(arg tf_prefix)/$(arg camera)_rgb_frame $(arg tf_prefix)/$(arg camera)_rgb_optical_frame" />  ```

Nan-Orbbec3d-US, 2019-08-13:
@faizan-m Thank you for your recommendation. I've updated this.

closed 2019-08-13 by faizan-m