orbbec/ros_astra_camera #91: [camera/camera_rgb-2] process has died

hoonkai, 2019-08-28:
Hi

I'm opening a separate issue as I'm not sure if it's related to my other issue #90. After running `roslaunch astra_camera astrapro.launch` for 15 or so seconds, the `camera_node` process appears to die:

```
  4884526 INFO       Depth.YRes was successfully set.
  4884549 INFO       Setting Depth.Registration to 1...
  4884552 INFO       Property Depth.FirmwareRegistration was changed to 1.
  4884554 INFO       Property Depth.Registration was changed to 1.
  4884555 INFO       Depth.Registration was successfully set.
[camera/camera_rgb-2] process has died [pid 6665, exit code -11, cmd /home/mwell/catkin_ws/devel/lib/astra_camera/camera_node /camera/image_raw:=/camera/rgb/image_raw __name:=camera_rgb __log:=/home/mwell/.ros/log/56825412-c952-11e9-a639-001c42b1c2c9/camera-camera_rgb-2.log].
log file: /home/mwell/.ros/log/56825412-c952-11e9-a639-001c42b1c2c9/camera-camera_rgb-2*.log
```

I've already followed the instructions closely. I'm running Ubuntu 18.04, melodic and using an Astra Pro.

Thanks

Sayter99, 2019-08-29:
Before crashing, could you see the image from `/camera/rgb/image_raw` topic?

hoonkai, 2019-08-29:
> Before crashing, could you see the image from `/camera/rgb/image_raw` topic?

Yes, I can. Sometimes it crashes after 10mins+ but sometimes just a few seconds.

> And could you provide your output of `lsusb` after plugging your Astra Pro. Thanks.

Sure. Here it is:

```
Bus 001 Device 008: ID 2bc5:0501  
Bus 001 Device 007: ID 2bc5:0403  
Bus 001 Device 003: ID 203a:fffa  
Bus 001 Device 002: ID 203a:fffa  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

Sayter99, 2019-08-29:
According to my experience, it's probably the USB connection issue. Could you try another USB cable and another USB slot? Thanks. #83 you can take a look at this issue for more information.

Sayter99, 2019-08-30:
And you can also try to switch the image codec from `yuyv` to `mjpeg` in `astrapro.launch`.

hoonkai, 2019-09-08:
> According to my experience, it's probably the USB connection issue. Could you try another USB cable and another USB slot? Thanks. #83 you can take a look at this issue for more information.

I've already tried a quality USB 2.0 cable and a quality USB 3.0 cable. It still crashes.

hoonkai, 2019-09-08:
> And you can also try to switch the image codec from `yuyv` to `mjpeg` in `astrapro.launch`.

If I change it to `mjpeg`, I get errors like below:

```
Corrupt JPEG data: 78 extraneous bytes before marker 0xd9
Corrupt JPEG data: 90 extraneous bytes before marker 0xd9
Corrupt JPEG data: 77 extraneous bytes before marker 0xd9
Premature end of JPEG file
Not a JPEG file: starts with 0x0e 0x32
Couldn't convert frame to RGB: Unknown error (-99)
Not a JPEG file: starts with 0x03 0x11
Couldn't convert frame to RGB: Unknown error (-99)
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 148 extraneous bytes before marker 0xd9

```

Sayter99, 2019-09-08:
Ok, let's try multiple things
1. Could you try latest version (I've updated it few days ago)
2. launch only RGB camera (comment out the other part)
3. enlarge your USB buffer by editing the file `/sys/module/usbcore/parameters/usbfs_memory_mb`. You can see a number in this file, probably it is 16 now. Try 64 if your system has enough memory.
Last, could you give me the serial number of your Astra Pro? Thank you.

hoonkai, 2019-09-09:
@Sayter99 I wonder if it's because my calibration files are not found. I'm using the stock `astrapro.launch` which contains

```
  <arg name="rgb_camera_info_url"   default="" />
  <arg name="depth_camera_info_url" default="" />
```

But is the rgb calibration file supposed to be named `rgb_[serial].yaml`? Is there a way to see if the file is loaded properly on launch though?

Sayter99, 2019-09-09:
It should be fine if you don't have calibration files. The program will use either the calibration parameters in your device or a calculated default one. You can echo the camera_info topics to make sure the program set parameters correctly. So, do the rgb node keep crashing after you tried those three things?

hoonkai, 2019-09-09:
Strange. `depth_<serial>.yaml` doesn't seem to work but `depth_Astra_Orbbec.yaml` does. Neither `rgb_<serial>.yaml` nor `rgb_Astra_Orbbec.yaml` works though.

Sayter99, 2019-09-10:
The calibration file is not well documented now. However, you can check the intrinsics and extrinsics by echoing /camera/*/camera_info. Even the calibration parameters are not loaded correctly, you still should be able to have correct raw images published by the node.

hoonkai, 2019-09-12:
@Sayter99 Using `rgb_<serial>.yaml`, `camera/rgb/camera_info` returns 0s. Can I ask what the expected filename should be? Like I mentioned, `depth_<serial>.yaml` works for the depth camera though.

Sayter99, 2019-09-12:
@hoonkai I think it should be another issue. It would be better to open another issue next time.

For RGB, because Astra Pro's RGB camera is a UVC, it controls by libuvc. So, you need to specify the file name in the launch file you use. For example,
```xml
<!--In node camera_node-->
<param name="camera_info_url" value="file:///tmp/cam.yaml"/>
```
You can change the path to what you want.

Last, are you able to fix the problem now?

hoonkai, 2019-09-17:
I've fixed it now. I had to use another USB port.

closed 2019-09-17 by hoonkai