orbbec/ros_astra_camera #94: No RGB or IR data published using Astra

VorpalBlade, 2019-10-03:
I'm using a non-pro, standard Astra.

Using the latest version of this repository on Ubuntu 16.04 with ROS Kinetic, only the depth data is published. No IR or RGB data is published to /camera/rgb/image_rect_color, /camera/ir/image etc.

I have checked with the sample tools in the Astra SDK, and there I can get RGB data, so it does not appear to be a hardware issue.

I have tried building with both catkin_tools (catkin build) and the traditional catkin_make.

An interesting message does show up in the console repeatedly at the end, indicating 0 FPS IR:

```
$ roslaunch astra_camera astra.launch
... logging to /home/arvid16/.ros/log/9f791dd2-e5a8-11e9-b3f8-8c1645fddd88/roslaunch-athena-7634.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://athena:46337/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/driver/auto_exposure: True
 * /camera/driver/auto_white_balance: True
 * /camera/driver/bootorder: 0
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: False
 * /camera/driver/device_id: #1
 * /camera/driver/devnums: 1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    driver (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
  /
    camera_base_link (tf2_ros/static_transform_publisher)
    camera_base_link1 (tf2_ros/static_transform_publisher)
    camera_base_link2 (tf2_ros/static_transform_publisher)
    camera_base_link3 (tf2_ros/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[camera/camera_nodelet_manager-1]: started with pid [7651]
process[camera/driver-2]: started with pid [7652]
process[camera/rgb_rectify_color-3]: started with pid [7653]
process[camera/depth_rectify_depth-4]: started with pid [7660]
process[camera/depth_metric_rect-5]: started with pid [7674]
process[camera/depth_metric-6]: started with pid [7690]
[ INFO] [1570085235.931100517]: Initializing nodelet with 4 worker threads.
process[camera/depth_points-7]: started with pid [7710]
process[camera/register_depth_rgb-8]: started with pid [7740]
process[camera/points_xyzrgb_sw_registered-9]: started with pid [7752]
process[camera/depth_registered_sw_metric_rect-10]: started with pid [7772]
[... 263 lines cut ...]
  7007793 VERBOSE    [FPS] IR: 0.00 Depth: 30.44 
  8008474 VERBOSE    [FPS] IR: 0.00 Depth: 30.32 
  9005997 VERBOSE    [FPS] IR: 0.00 Depth: 30.33 
 10008419 VERBOSE    [FPS] IR: 0.00 Depth: 30.33 
 11018378 VERBOSE    [FPS] IR: 0.00 Depth: 30.25 
 12004056 VERBOSE    [FPS] IR: 0.00 Depth: 30.35 
 13005914 VERBOSE    [FPS] IR: 0.00 Depth: 30.36 
 14006558 VERBOSE    [FPS] IR: 0.00 Depth: 30.45 
 15006948 VERBOSE    [FPS] IR: 0.00 Depth: 30.31 
 16010223 VERBOSE    [FPS] IR: 0.00 Depth: 30.42 
 17009151 VERBOSE    [FPS] IR: 0.00 Depth: 30.31 
 18015780 VERBOSE    [FPS] IR: 0.00 Depth: 30.27 
 19003645 VERBOSE    [FPS] IR: 0.00 Depth: 30.40 
 20019283 VERBOSE    [FPS] IR: 0.00 Depth: 30.24 
 21012271 VERBOSE    [FPS] IR: 0.00 Depth: 30.37 
 22021125 VERBOSE    [FPS] IR: 0.00 Depth: 30.16 
 23008688 VERBOSE    [FPS] IR: 0.00 Depth: 30.44 
 24012686 VERBOSE    [FPS] IR: 0.00 Depth: 30.36 
 25010928 VERBOSE    [FPS] IR: 0.00 Depth: 30.44 
 26016345 VERBOSE    [FPS] IR: 0.00 Depth: 30.26 
 27007434 VERBOSE    [FPS] IR: 0.00 Depth: 30.39 
 28004932 VERBOSE    [FPS] IR: 0.00 Depth: 30.39 
 29012330 VERBOSE    [FPS] IR: 0.00 Depth: 30.37 
 30015117 VERBOSE    [FPS] IR: 0.00 Depth: 30.27 
```

VorpalBlade, 2019-10-03:
Whoops, seems like it is actually an Astra Pro and I was using the wrong launch file. The box was just labeled Astra, as was the manual, so I assumed it was a non-Pro.

The pro needs a different launch file. With that the RGB stream works, but the IR stream still doesn't.

Sayter99, 2019-10-10:
Does it work If you just subscribe `/camera/ir/image`?

VorpalBlade, 2019-10-10:
No, all I get is a completely black image.

If I'm also subscribing to the depth image at the same time I get a series message that I don't get when subscribing only to the IR image:

```
  8690741 WARNING    Depth: Expected 1348, got 134a
  8704135 WARNING    Read: Depth buffer is corrupt. Size is 204736 (!= 614400)
  8720009 WARNING    Depth: Expected 13ba, got 13c2
  8720045 WARNING    Depth frame is corrupt!
  8724130 WARNING    Depth: Expected 13d2, got 13d4
  8737567 WARNING    Read: Depth buffer is corrupt. Size is 204736 (!= 614400)
```

Note that RGB and depth images work fine, with no unexpected messages.

Sayter99, 2019-10-10:
It's normal to get messy signals while open three streams at the same time because of the bandwidth limitation. Are you using RViZ? The IR image is dark if you don't normalize it, however I remember rviz always shows normalized IR image. So you could give it a try. Or if you can see the FPS of IR stream, it means there is data streaming.

VorpalBlade, 2019-10-10:
I used rqt ImageView before. Now I tested with rviz. It is dark in both cases. However, rviz also displays a warning: "No image received".

Sayter99, 2019-10-10:
According to the log and error messages, I think the driver doesn't receive valid image data from IR stream. Can you receive images by Astra SDK or OpenNI SDK? Second, could you try to enlarge your usb buffer stating in /sys/module/usbcore/parameters/usbfs_memory_mb. Try to double this value and see if the errors get reduced.

VorpalBlade, 2019-10-11:
Increasing usbfs_memory_mb reduces the frequency of the error messages when multiple streams are present (but the errors are still not completely gone).

Looking at the samples in the Astra SDK I don't see any viewer for the raw IR data, only for the depth & colour data, unless that is what MultiSensorViewer-SFML provides. With that sample I get two views, one is colour data (top right), and one is a black and white image (top left) with a wavy pattern on it. I also get what appears to be space below for two more streams but where there is only blackness.

Sayter99, 2019-10-11:
Can you get any image now? Or only reduce error messages. Next, try niviewer in openni sdk provided by Orbbec to see if there is any IR image. In my perspective, this issue is about usb connection, bandwidth problems. You can also test on another computers or usb slots.

VorpalBlade, 2019-10-14:
I do not get an image even with the reduced errors, sorry if that wasn't clear. I have tried on two computers (a Lenovo ThinkPad T480 and an Intel NUC).

However, using NiViewer I do seem to get a depth image and an IR image (a black and white image with a clear speckle pattern on it). I do not get the RGB image however. This is using OpenNI_2.3.0.63.zip downloaded from Orbbec's website (the x64 Linux build inside that zip).

Sayter99, 2019-10-14:
It's normal that you cannot get RGB images by OpenNI2. Because Astra Pro adopts UVC camera for RGB images, it is not integrated with OpenNI2. And this package is based on OpenNI2, so I utilize another package called libuvc to get RGB images from UVC. For now, I think it's a bandwidth issue as I mentioned before. If you want to use depth + RGB, it's okay now. If you want to use IR + depth, you need to comment out the libuvc section in the launch file (astra_camera part). If you need to see all of them, probably reducing the resolution is a way to try. Could you tell me more about your requirements?

VorpalBlade, 2019-10-15:
I was interested in getting IR data to be able to do intrinsic calibration (I was planning to compare the accuracy of hardware and software depth registration). In the end I need depth data, I even don't  care about the RGB data except for calibration purposes. In addition there is a weird pattern of "holes" in the hardware registered depth image, which I wanted to investigate by comparing it to software processing.

I tested disabling libuvc, but the image is still completely black in both rviz and rqt imageview for IR. I'm not using the depth stream at the same time. The IR emitter is turned on, and according to the console output of the driver, it is streaming:

    172006226 VERBOSE    [FPS] IR: 30.33 Depth: 0.00

Sayter99, 2019-10-15:
Can you try rostopic echo as well? For calibration, I think you can try to calibrate RGB and IR separately and then utilize the result to do registration.

VorpalBlade, 2019-10-15:
It appears that rostopic echo does indeed give data on /sensors/camera/ir/image! And it appears to contain image data (not just zeros).

Then the question is, why rviz, rqt and more importantly camera_calibration doesn't work with this IR stream. It seems particularly strange that rviz even claims "No image received".

Sayter99, 2019-10-15:
The problem is you need to do normalization before showing. IR image is a 16-bit image, to show this image clearly, you need to normalize it to (0-255) and be an 8-bit image. For the rviz, could you attach some screenshots? Rviz usually helps us normalize images and show.

VorpalBlade, 2019-10-16:
Ah, it appears it makes a difference if I select "Camera" or "image" in rviz as the display type to visualise the topic. With "Image" it works, with "Camera" it doesn't. Whoops, and good to know.

I guess I will need to write a normalisation nodelet to be able to use it with camera_calibration.

I would suggest that it should be documented somewhere in the README that the IR image is not normalised, to avoid others getting confused as well.

Thanks for all the support!

Sayter99, 2019-10-16:
Good idea. I will think about this. You're welcome.

VorpalBlade, 2019-10-17:
Should I close the issue or do you want it to track adding adding the note to the documentation?

closed 2019-10-18 by VorpalBlade