orbbec/ros_astra_camera #122: RGB Issues

starrche, 2020-08-20:
I am using the Orbbec Astra Pro with Ubuntu 18.04 and ROS Melodic.

When I run astra.launch I can then run rosrun image_view on /camera/depth/image, but when I try the IR topics or RGB topics I don't see anything. No error messages either. I have copied the results from running _roswtf_ both on the astra_camera package and from my home directory. I have libuvc installed, so I am unsure if that error is the issue.   

*nvidia@nvidia-Inspiron-7373:~/catkin_ws/src/ros_astra_camera/include$ roswtf
Loaded plugin tf.tfwtf
Package: astra_camera
Cannot find dependencies for package [astra_camera]: missing libuvc
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/nvidia/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share

nvidia@nvidia-Inspiron-7373:~$ roswtf
Loaded plugin tf.tfwtf
No package or stack in the current directory
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules

Online checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /camera/camera_nodelet_manager:
   * /camera/rgb/camera_info
   * /camera/rgb/image_raw*

starrche, 2020-08-20:
Additionally, I have tried the astrapro.launch file, but it fails immediately with the following error:

```
[camera/camera_rgb-2] process has died [pid 19952, exit code 255, cmd /home/nvidia/catkin_ws/devel/lib/astra_camera/camera_node /camera/image_raw:=/camera/rgb/image_raw __name:=camera_rgb __log:=/home/nvidia/.ros/log/f0c5e520-e314-11ea-9ce9-44032cfc753d/camera-camera_rgb-2.log].
log file: /home/nvidia/.ros/log/f0c5e520-e314-11ea-9ce9-44032cfc753d/camera-camera_rgb-2*.log

```

label01, 2020-09-04:
me too. Did you slove it?

starrche, 2020-09-05:
The astra.launch file is not the correct launch file for the Astra Pro. The astrapro.launch file needs to be used. Unfortunately, I am also getting error with the astrapro launch file. At least astra.launch gives me depth data. astrapro.launch fails right away with the same error as issue #91. I haven't solved it yet.

label01, 2020-09-07:
Thank you.
I have fixed it.
--------------------------------------------------------------------------------------------------------------------
$lsusb
2bc5:0502 is the rgb camera.
$vim astrapro.launch

catkin_make again.

starrche, 2020-09-09:
Hmmm that didn't fix it for me. I made sure astrapro.launch was referring to the correct device and double checked the udev rules.

ChowhoundWu, 2020-12-28:
It occurs to me,too,ROS melodic
The depth image and the ir image is right,but can not get the rgb image.
The error after I roslaunch astrapro.launch is the follows.
camera/camera_rgb-2] process has died