orbbec/ros_astra_camera #123: Memory leak on the openni:open()

alexfneves, 2020-08-28:
I'm doing some changes to add a reconnection feature on the ROS node, and I noticed a memory leak on the `openni::open()` method. This is happening on the:
https://github.com/orbbec/ros_astra_camera/blob/05ce57b20bfe07d82c447f22a069f6bf3e630d7d/src/astra_device_manager.cpp#L225
The memory leak is seen when I start the node with a specific serial number, and there is an astra camera on the computer with another serial number, so the code will go through the `getSerial()`, the `openni_device.open(Uri.c_str())` will open the device, grab the serial number and return the function. The openni_device destructor will be called.

I saw the memory leak on that function while using heaptrack. Since the openni library is already precompiled, I can't debug the issue any further. I don't have the heaptrack anymore, but later I can generate it again and show here.

brinij, 2020-11-27:
Hi, thank you for posting this! Did you solve this so far?

alexfneves, 2020-11-27:
Hi. No, since the openni library comes compiled I couldn't debug any further. And I also don't have much time to debug any further, I'm just living with the bug.

brinij, 2020-11-27:
Okay. I am asking because I have noticed that when running two Embedded S Astra for a longer time there is a memory leak, but with two Astra Mini S there is no. So I am not sure if it is related to this.