orbbec/ros_astra_camera #131: AstraPro not using the new RGB calibration file

YamenMh, 2021-01-12:
Hey everyone,

so before calibrating i was getting warnings from **astrapro.launch** that **camera.yaml** and **depth_Astra_Orbbec.yaml**.
After calibrating by using the camera_calibration package (http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) I cereated two files:  **depth_Astra_Orbbec.yaml** and  **rgb_Astra_Orbbec.yaml**

The  **depth_Astra_Orbbec.yaml** is being used and one can see the values when calling **rostopic camera/depth/camera_info**, but thats not the case with **RGB**.  **rostopic camera/rgb/camera_info** is showing another values as in **rgb_Astra_Orbbec.yaml**.

I tried all the following:
-astrapro.launch was still showing the warning that camera.yaml missing ---> I made a copy  of **rgb_Astra_Orbbec.yaml** and called it **camera.yaml** ----> warning disappeard but the values in **rostopic camera/rgb/camera_info** did not change

-**rosservice call /camera/get_camera_info** showed the same values as **rostopic camera/rgb/camera_info** ---> i tried to cahnge the calibration data manually with **rosservice call /camera/set_camera_info** --> this command cereates a **camera.yaml**  file but that also did not change anything  in the showed calibration values

-in **astrapro.launch** there are the following lines to refer to the calibration files 
  **arg name="rgb_camera_info_url"   default=""**  ---> refering to **rgb_Astra_Orbbec.yaml** did not make a difference
  **arg name="depth_camera_info_url" default=""** ---> refering to depth calibration file works
  **param name="camera_info_url" value=""**  this one should be the url to the **camera.yaml** file ---> cereating the **camera.yaml** and linking to it did not cahnge anything

In the terminal of astrapro.launch I can see:
10104987 VERBOSE    Multi distance calibration enable 0
 10104990 VERBOSE    =================================================
 10105013 VERBOSE    **depth** [577.530334,577.530334,316.777740,239.781937], 
**color**[597.587158,597.587158,318.318207,245.468109]

These values are being used for depth and rgb. As I said refering to the depth calibration file works and changes the values but even after refering to rgb or camera yaml file the values from above **(color)** are still being used as RGB calibration values.

I appreciate any help :)

omair-khalid, 2021-03-25:
Try using a format like this:  ` file://${ROS_HOME}/camera_info/depth_Astra_Orbbec.yaml`

YamenMh, 2021-03-25:
Yes, this is working for the depth_Astra_Orbbec.yaml, but not for the rgb_Astra_Orbbec.yaml as mentioned above

omair-khalid, 2021-03-25:
As for me, I am using the ros_astra_launch package given by clearpathrobotics. They use a launch file called astra_pro.launch. Using this combination, I was able to load both rgb and depth calibration files. 

`
<launch>

  
  <arg name="camera" default="camera" />
  <arg name="rgb_frame_id"   default="$(arg camera)_rgb_optical_frame" />
  <arg name="depth_frame_id" default="$(arg camera)_depth_optical_frame" />

  
  <arg name="device_id" default="#1" />
  <arg name="bootorder" default="0" />
  <arg name="devnums" default="1" />

  
  <arg name="rgb_camera_info_url"   default="file://${ROS_HOME}/camera_info/rgb_Astra_Orbbec.yaml" />
  <arg name="depth_camera_info_url" default="file://${ROS_HOME}/camera_info/depth_Astra_Orbbec.yaml" />

  
  <arg name="depth_registration" default="false" />

  
  <arg name="color_depth_synchronization"     default="false" />
  <arg name="auto_exposure"                   default="true" />
  <arg name="auto_white_balance"              default="true" />

  
  <arg name="rgb"              default="rgb" />
  <arg name="ir"               default="ir" />
  <arg name="depth"            default="depth" />

  
  <arg name="load_driver" default="true" />
  <arg name="publish_tf" default="true" />
  
  <arg name="rgb_processing"                  default="true"  />
  <arg name="debayer_processing"              default="true" />
  <arg name="ir_processing"                   default="false" />
  <arg name="depth_processing"                default="true" />
  <arg name="depth_registered_processing"     default="true" />
  <arg name="disparity_processing"            default="false" />
  <arg name="disparity_registered_processing" default="false" />
  <arg name="hw_registered_processing"        default="false" if="$(arg depth_registration)" />
  <arg name="sw_registered_processing"        default="true" if="$(arg depth_registration)" />
  <arg name="hw_registered_processing"        default="false" unless="$(arg depth_registration)" />
  <arg name="sw_registered_processing"        default="true" unless="$(arg depth_registration)" />

  
  <arg name="respawn" default="false" />

  
  <arg name="num_worker_threads" default="4" />

  
  <group ns="$(arg camera)">

    
    <arg name="manager" value="$(arg camera)_nodelet_manager" />
    <arg name="debug" default="false" /> 
    <include file="$(find rgbd_launch)/launch/includes/manager.launch.xml">
      <arg name="name" value="$(arg manager)" />
      <arg name="debug" value="$(arg debug)" />
      <arg name="num_worker_threads"  value="$(arg num_worker_threads)" />
    </include>

    
    <include if="$(arg load_driver)"
       file="$(find astra_launch)/launch/includes/device.launch.xml">
      <arg name="manager"                         value="$(arg manager)" />
[... 47 lines cut ...]
        <param name="height" value="480"/>
        
        <param name="video_mode" value="uncompressed"/> 
        <param name="frame_rate" value="30"/>
        <param name="frame_id" value="$(arg camera)_rgb_optical_frame"/>

        <param name="timestamp_method" value="start"/> 
        <param name="camera_info_url" value="file://${ROS_HOME}/camera_info/rgb_Astra_Orbbec.yaml"/>

        <param name="auto_exposure" value="3"/> 
        <param name="auto_white_balance" value="false"/>
      </node>

    </group> 

  </group> 

  
  <include if="$(arg publish_tf)"
     file="$(find astra_launch)/launch/includes/astra_frames.launch">
    <arg name="camera" value="$(arg camera)" />
  </include>

</launch>
`

YamenMh, 2021-03-31:
That worked for me, thank you a lot!  Using the libuvc_camera package in astrapro.launch instead of the astra_camera package solved it.