orbbec/ros_astra_camera #132: USB Connection and hanging issues with Astra

jaymwong, 2021-01-21:
We've been witnessing several issues running the Astra (non-Pro) cameras with the ROS driver long term. Has anyone had success with running these cameras in production 24/7 for multiple days straight? 

The largest barriers are for our team appear to be,
- **USB Connection issue:** (related to https://github.com/orbbec/astra/issues/28), where the connection fail and is impossible to recover from; we've tried using various methods to automatically reset the USB. The only solution here is to do a `sudo reboot` on the linux machine the camera is connected to. 
-  **Initial bootup hanging**. Periodically the camera will be detected, but will hang with the following output (neither RGB nor IR images will be published) and the fix is another `sudo reboot`

```
  3866620 INFO       Property Depth.Registration was changed to 1.
  3866622 INFO       Depth.Registration was successfully set.
  3866635 INFO       Setting Image.AutoExposure to 1...
  3866637 INFO       Image.AutoExposure value did not change.
  3866644 INFO       Setting Image.AutoWhiteBalance to 1...
  3866646 INFO       Image.AutoWhiteBalance value did not change.
```
  
Any help regarding these issues or any work-arounds would be greatly appreciated!

brinij, 2021-01-25:
How are you connecting your cameras to the PC? Do you use USB extension cables?

jaymwong, 2021-01-25:
Yes we are connecting the Astra to the PC. We're able to obtain the RGB, IR images, and point clouds fine, but periodically these issues arise making long term deployment problematic.

We are using a USB3.0 extension cable with a active repeater/signal booster because the PC and the desired mounting area for the camera is about 50ft away (mounted 3+ meters high). Requiring the Astra to be connected to the PC directly without extensions is an unsatisfying answer because and accessibility to the human operator makes this infeasible.

brinij, 2021-01-25:
We are experiencing the same issue. In our system, Astra Mini S is connected with a 3m long USB 2.0 extension cable.

jaymwong, 2021-06-09:
**Update:** Since this is still an unsolved problem, we have decided to swap every single Astra camera in our system with Intel Realsense cameras. Unfortunately it doesn't seem like we can not work around this reliability issue; it's a shame since we really like the Astra cameras. 

If anyone finds a solve, our team would be happy to revisit these cameras one day.