orbbec/ros_astra_camera #138: Run astrapro.launch RGB Topic exit abnormally

mailonghua, 2021-04-14:
I am using the Orbbec Astra Pro with Ubuntu 18.04 and ROS Melodic.
When I run astrapro.launch
I can see the depth image but cannot see the rgb image, the corresponding topic will exit abnormally
error log:

roslaunch astra_camera astrapro.launch

[camera/camera_rgb-2] process has died [pid 39876, exit code -11, cmd /home/mailonghua/catkin_ws/devel
/lib/astra_camera/camera_node /camera/image_raw:=/camera/rgb/image_raw __name:=camera_rgb __log:=/home
/mailonghua/.ros/log/562777f2-9d33-11eb-bb14-000c29699901/camera-camera_rgb-2.log].                   
log file: /home/mailonghua/.ros/log/562777f2-9d33-11eb-bb14-000c29699901/camera-camera_rgb-2*.log

mailonghua, 2021-04-15:
[roslaunch][INFO] 2021-04-15 01:09:06,632: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2021-04-15 01:09:06,637: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2021-04-15 01:09:06,637: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'astra_camera', 'astrapro.launch']
[roslaunch][INFO] 2021-04-15 01:09:06,637: roslaunch env is {'ROS_DISTRO': 'melodic', 'LESS': '-R', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'GJS_DEBUG_OUTPUT': 'stderr', 'WINDOWPATH': '2', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'XDG_SESSION_TYPE': 'x11', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/mailonghua/catkin_ws/devel/share/common-lisp', 'QT_IM_MODULE': 'ibus', 'LOGNAME': 'mailonghua', 'USER': 'mailonghua', 'PATH': '/opt/ros/melodic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'XDG_VTNR': '2', 'HOME': '/home/mailonghua', 'CMAKE_PREFIX_PATH': '/home/mailonghua/catkin_ws/devel:/opt/ros/melodic', 'ZSH': '/home/mailonghua/.oh-my-zsh', 'DISPLAY': ':0', 'SSH_AGENT_PID': '2371', 'LANG': 'en_US.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/zsh', 'ROS_LOG_FILENAME': '/home/mailonghua/.ros/log/d95b69f6-9dc1-11eb-a185-000c29699901/roslaunch-ubuntu-5173.log', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'SESSION_MANAGER': 'local/ubuntu:@/tmp/.ICE-unix/2245,unix/ubuntu:/tmp/.ICE-unix/2245', 'SHLVL': '2', 'QT_ACCESSIBILITY': '1', 'LD_LIBRARY_PATH': '/home/mailonghua/catkin_ws/devel/lib:/opt/ros/melodic/lib:/opt/ros/melodic/lib/x86_64-linux-gnu', 'QT4_IM_MODULE': 'xim', 'CLUTTER_IM_MODULE': 'xim', 'TEXTDOMAIN': 'im-config', 'IM_CONFIG_PHASE': '2', 'ROS_MASTER_URI': 'http://localhost:11311', 'XDG_RUNTIME_DIR': '/run/user/1000', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'ROS_PYTHON_VERSION': '2', 'USERNAME': 'mailonghua', 'XDG_SESSION_DESKTOP': 'ubuntu', 'ROS_VERSION': '1', 'PYTHONPATH': '/home/mailonghua/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'VTE_VERSION': '5202', 'TMUX': '/tmp/tmux-1000/default,4274,0', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'GDMSESSION': 'ubuntu', 'XMODIFIERS': '@im=ibus', 'TEXTDOMAINDIR': '/usr/share/locale/', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'PKG_CONFIG_PATH': '/home/mailonghua/catkin_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig', 'ROS_PACKAGE_PATH': '/home/mailonghua/catkin_ws/src:/opt/ros/melodic/share', 'XDG_SESSION_ID': '2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', '_': '/opt/ros/melodic/bin/roslaunch', 'GTK_MODULES': 'gail:atk-bridge', 'GTK_IM_MODULE': 'ibus', 'DESKTOP_SESSION': 'ubuntu', 'LSCOLORS': 'Gxfxcxdxbxegedabagacad', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'GNOME_TERMINAL_SERVICE': ':1.70', 'XDG_SEAT': 'seat0', 'OLDPWD': '/home/mailonghua/catkin_ws/src/ros_astra_camera', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'PWD': '/home/mailonghua/catkin_ws/src/ros_astra_camera', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'COLORTERM': 'truecolor', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'PAGER': 'less', 'GJS_DEBUG_TOPICS': 'JS ERROR;JS LOG', 'TMUX_PANE': '%0', 'GNOME_TERMINAL_SCREEN': '/org/gnome/Terminal/screen/f19aac1a_5cd7_4173_ac1d_7be34f492947'}
[roslaunch][INFO] 2021-04-15 01:09:06,638: starting in server mode
[roslaunch.parent][INFO] 2021-04-15 01:09:06,638: starting roslaunch parent run
[roslaunch][INFO] 2021-04-15 01:09:06,638: loading roscore config file /opt/ros/melodic/etc/ros/roscore.xml
[roslaunch][INFO] 2021-04-15 01:09:07,137: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2021-04-15 01:09:07,137: loading config file /home/mailonghua/catkin_ws/src/ros_astra_camera/launch/astrapro.launch
[roslaunch][INFO] 2021-04-15 01:09:07,143: Added node of type [astra_camera/camera_node] in namespace [/camera/]
[... cut ...]
[roslaunch][ERROR] 2021-04-15 01:09:10,874: [camera/camera_rgb-2] process has died [pid 5201, exit code -11, cmd /home/mailonghua/catkin_ws/devel/lib/astra_camera/camera_node /camera/image_raw:=/camera/rgb/image_raw __name:=camera_rgb __log:=/home/mailonghua/.ros/log/d95b69f6-9dc1-11eb-a185-000c29699901/camera-camera_rgb-2.log].
log file: /home/mailonghua/.ros/log/d95b69f6-9dc1-11eb-a185-000c29699901/camera-camera_rgb-2*.log
[roslaunch.pmon][INFO] 2021-04-15 01:09:10,874: ProcessMonitor.unregister[camera/camera_rgb-2] starting
[roslaunch.pmon][INFO] 2021-04-15 01:09:10,874: ProcessMonitor.unregister[camera/camera_rgb-2] complete

c-starr, 2021-06-01:
I am having the same issue. Did you figure it out?

c-starr, 2021-06-01:
@mailonghua Changing the device ID to the following values fixed it for me. Lines 70 & 71

    param name="vendor" value="0x6366"
    param name="product" value="0x0c45" 

The previous values related to the depth camera. I believe the values above reflect the rgb camera.