orbbec/ros_astra_camera #140: Problem connecting several Astra Pro Cameras

danifpdra, 2021-06-01:
Hi,

I am trying to connect several Astra Cameras to the same workstation. I connected the cameras to different USB ports and when I run the multi_astra.launch file or several astrapro.launch files, I always get the error 

`[ERROR] [1622536758.125510292]: Can't open /dev/bus/usb/001/032: Busy (-6)`

Did anyone experience the same problem? I have tried to connect two devices in different USB ports in my personal laptop and it works, but in the workstation, the exact same procedure produces that error.

@miguelriemoliveira

dgcnz, 2021-11-26:
I'm experiencing the same problem with 2 Astra Embedded S Cameras, were you able to fix it? @danifpdra

dgcnz, 2021-12-02:
My problem was solved by disabling uvc_serial and uncommenting the `<arg name="index" ...>` for both cameras. :)