orbbec/ros_astra_camera #148: AstraPro cannot stream rgb and depth simultaneously

liver121888, 2021-12-30:
I was trying to use AstraPro for Rtab-map, I run the following command and get the error

> roslaunch astra_launch astrapromodi.launch

> [ INFO] [1640876066.260598811]: device name: Orbbec Astra Pro
unsupported descriptor subtype VS_COLORFORMAT
attempt to claim already-claimed interface 1
[ WARN] [1640876067.178280454]: Unable to set scanning_mode to 0
[ WARN] [1640876067.179600022]: Unable to set exposure_absolute to 1
[ WARN] [1640876067.180068212]: Unable to set auto_focus to 1
[ WARN] [1640876067.181163659]: Unable to set pantilt to -648000, 127

The depth image works fine, but the RGB image does not work and the process is died.

My launch file: 

>   <node pkg="astra_camera" type="camera_node" name="$(arg camera)_rgb">
      
      <param name="vendor" value="0x2bc5"/>
      <param name="product" value="0x0502"/>
      <param name="serial" value="0"/>
      
      <param name="index" value="$(arg index)"/>

      
      <param name="width" value="640"/>
      <param name="height" value="480"/>
      
      <param name="video_mode" value="yuyv"/> 
      <param name="frame_rate" value="30"/>

      <param name="timestamp_method" value="start"/> 
      <param name="camera_info_url" value=""/>

       
      
      <remap from="/$(arg camera)/image_raw" to="/$(arg camera)/rgb/image_raw" />
    </node>

So I found out that if I change yuyv to mjpeg, the RGB works but the depth fails.

I am sure the USB device number is right, and I try to launch a node inside the same launch file using usb_cam, but because they are on the same USB, the resource is occupied. 
I found that a lot of people encountered the same problem, but none of them actually solve my problem.

izisKanjaraw, 2023-12-19:
@liver121888  hi , please can u tell me if u could fix the error cause i am facing the same

liver121888, 2024-01-01:
Sorry, I didn't fix the error. I changed from AstraPro to RealSense 435i.

closed 2024-11-05 by liver121888