orbbec/ros_astra_camera #149: Broken Depth Stream on Astra Embedded S after 12/2021 commits

dgcnz, 2022-01-15:
Running the embedded_s.launch on the last commit (aff227d64e226166315c2afe0d7f4a5b1c99253d, at the time of writing), yields a funky point cloud.

[image]

However, reverting back to 78186a3ffbf8a67cfdf409abeebc68c402b406a0 seems to do the trick.
[image]

Specifically, be006e1760c1442e53da6621cd74a88bfe606c2d is the first commit where my program stops working correctly.

ingjae, 2022-08-03:
I have same issue

e-dul-xb, 2022-08-26:
Set depth video mode to 640x400, this will match device spec and give you valid depth.
btw. after refactor some sample launch files doesn't work anymore :sweat_smile: it's not helping to evaluate such product