orbbec/ros_astra_camera #153: Access denied (insufficient permissions)!

1982582704, 2022-06-20:
[ INFO] [1655719122.489360808]: No matching device found.... waiting for devices. Reason: astra_wrapper::AstraDevice::AstraDevice(const string&) @ /home/he/catkin/src/ros_astra_camera-master/src/astra_device.cpp @ 79 : Device open failed
	Could not open "2bc5/0404@2/17": Access denied (insufficient permissions)!

[image]

jian-dong, 2022-07-26:
May be you missing copy udev rules to system, Could you please try
```bash
sudo cp 56-orbbec-usb.rules /etc/udev/rules.d
sudo udevadm control --reload-rules && sudo udevadm trigger
```

an99990, 2022-09-08:
i have the same issue
tried your solution i got : 
`cp: cannot stat '56-orbbec-usb.rules /etc/udev/rules.d': No such file or directory`

an99990, 2022-09-08:
after i did the second line i got another issue :

`Unsupported IR video mode - Resolution: 640z480@15Hz Format: Gray16`