orbbec/ros_astra_camera #160: Orbbec Stereo S not detected by ROS driver

mbs-tahir, 2022-08-04:
I have an Orbbec Stereo S camera which I am intended to use with ROS. The camera is detected(`lsusb`), udev rule is working (can see 
 `/dev/astra`, `/dev/astrauvc`), by using third party software I can also the RGB stream of the camera.

`lsusb` output:
```
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 008: ID 5986:014c Acer, Inc BisonCam, NB Pro
Bus 003 Device 005: ID 13d3:3394 IMC Networks Bluetooth
Bus 003 Device 022: ID 062a:4101 MosArt Semiconductor Corp. Wireless Keyboard/Mouse
Bus 003 Device 036: ID 2bc5:0508  USB2.0 Hub
Bus 003 Device 035: ID 2bc5:0608 GenesysLogic USB2.1 Hub
Bus 003 Device 034: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 010: ID 06c4:c411 Bizlink International Corp. USB2.1 Hub
Bus 003 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 011: ID 062a:4101 MosArt Semiconductor Corp. Wireless Keyboard/Mouse
Bus 003 Device 007: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 009: ID 17e9:6006 DisplayLink USB2.1 Hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
for UDEV rules:
```
mbs@Development:~$ ls /dev/a*
/dev/astra  /dev/astrauvc  /dev/autofs

```

ROS driver installation is working fine but when I run the script to find devices: 

```
mbs@Development:~$ rosrun astra_camera astra_list_devices 
Found 0 devices:
```
Similarly when running the ROS driver it says:
```
mbs@Development:~$ roslaunch astra_camera stereo_s.launch 
... logging to /home/mbs/.ros/log/83269ac6-13d9-11ed-a7e2-573c93823650/roslaunch-Development-92335.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Development:33587/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /camera/driver/auto_exposure: True
 * /camera/driver/auto_white_balance: True
 * /camera/driver/bootorder: 0
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: True
 * /camera/driver/device_id: #1
 * /camera/driver/devnums: 1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /camera/stereo_s/camera_info_url: 
 * /camera/stereo_s/frame_rate: 30
 * /camera/stereo_s/height: 480
 * /camera/stereo_s/index: 0
 * /camera/stereo_s/product: 0x0508
 * /camera/stereo_s/serial: 0
 * /camera/stereo_s/timestamp_method: start
 * /camera/stereo_s/vendor: 0x2bc5
 * /camera/stereo_s/video_mode: mjpeg
 * /camera/stereo_s/width: 640
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    camera_base_link (tf2_ros/static_transform_publisher)
    camera_base_link1 (tf2_ros/static_transform_publisher)
    camera_base_link2 (tf2_ros/static_transform_publisher)
    camera_base_link3 (tf2_ros/static_transform_publisher)
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    driver (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    stereo_s (astra_camera/camera_node)

ROS_MASTER_URI=http://localhost:11311

process[camera/stereo_s-1]: started with pid [92349]
process[camera/camera_nodelet_manager-2]: started with pid [92350]
process[camera/driver-3]: started with pid [92351]
process[camera/rgb_rectify_color-4]: started with pid [92353]
process[camera/depth_rectify_depth-5]: started with pid [92359]
process[camera/depth_metric_rect-6]: started with pid [92364]
process[camera/depth_metric-7]: started with pid [92370]
process[camera/depth_points-8]: started with pid [92376]
process[camera/depth_registered_rectify_depth-9]: started with pid [92379]
process[camera/points_xyzrgb_hw_registered-10]: started with pid [92386]
process[camera/depth_registered_hw_metric_rect-11]: started with pid [92389]
process[camera/depth_registered_metric-12]: started with pid [92396]
process[camera_base_link-13]: started with pid [92408]
process[camera_base_link1-14]: started with pid [92410]
process[camera_base_link2-15]: started with pid [92418]
process[camera_base_link3-16]: started with pid [92423]
[ INFO] [1659610945.261853722]: Initializing nodelet with 4 worker threads.
[ WARN] [1659610945.958396366]: ~device_id :#1
[ INFO] [1659610945.970165712]: bootOrder: 0, devnums: 1, device_id: #1
[ INFO] [1659610945.970604020]: No matching device found.... waiting for devices. Reason: std::string astra_wrapper::AstraDriver::resolveDeviceURI(const string&) @ /home/mbs/catkin_ws/src/astra_camera/src/astra_driver.cpp @ 1245 : Invalid device number 1, there are 0 devices connected.
[ INFO] [1659610948.971044699]: No matching device found.... waiting for devices. Reason: std::string astra_wrapper::AstraDriver::resolveDeviceURI(const string&) @ /home/mbs/catkin_ws/src/astra_camera/src/astra_driver.cpp @ 1245 : Invalid device number 1, there are 0 devices connected.
[ INFO] [1659610951.971248900]: No matching device found.... waiting for devices. Reason: std::string astra_wrapper::AstraDriver::resolveDeviceURI(const string&) @ /home/mbs/catkin_ws/src/astra_camera/src/astra_driver.cpp @ 1245 : Invalid device number 1, there are 0 devices connected.
^C[camera_base_link3-16] killing on exit
[camera_base_link2-15] killing on exit
[camera_base_link1-14] killing on exit
[camera_base_link-13] killing on exit
[camera/depth_registered_metric-12] killing on exit
[camera/depth_metric-7] killing on exit
[camera/depth_registered_hw_metric_rect-11] killing on exit
[camera/points_xyzrgb_hw_registered-10] killing on exit
[camera/depth_registered_rectify_depth-9] killing on exit
[camera/depth_points-8] killing on exit
[camera/depth_metric_rect-6] killing on exit
[camera/depth_rectify_depth-5] killing on exit
[camera/rgb_rectify_color-4] killing on exit
[camera/driver-3] killing on exit
[camera/camera_nodelet_manager-2] killing on exit
[camera/stereo_s-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
Any help will really be appreciated as I am not able to understand where the problem is either on software side or on hardware side?

mbs-tahir, 2022-08-08:
udev rules are there I double checked. May be you can confirm from: 

```
mbs@Development:~$ ls /dev/a*
/dev/astra  /dev/astrauvc  /dev/autofs
```

jian-dong, 2022-08-16:
Hi @mbs-tahir , I finally reproduced this issue in rk3399, Can you try using USB2.0 instead of USB3.0, In my case it works.