orbbec/ros_astra_camera #161: Trouble building package

PhilipAmadasun, 2022-08-05:
I get:
```
Starting  >>> astra_camera                                            
______________________________________________________________________
Errors     << astra_camera:make /home/turtlebot/astra_ws/logs/astra_camera/build.make.000.log
In file included from /opt/ros/kinetic/include/ros/assert.h:35:0,
                 from /opt/ros/kinetic/include/ros/common.h:36,
                 from /opt/ros/kinetic/include/ros/publisher.h:32,
                 from /opt/ros/kinetic/include/ros/node_handle.h:32,
                 from /home/turtlebot/astra_ws/devel/.private/astra_camera/include/astra_camera/AstraConfig.h:21,
                 from /home/turtlebot/astra_ws/src/ros_astra_camera/include/astra_camera/astra_driver.h:36,
                 from /home/turtlebot/astra_ws/src/ros_astra_camera/src/astra_device.cpp:42:
/home/turtlebot/astra_ws/src/ros_astra_camera/src/astra_device.cpp: In constructor ‘astra_wrapper::AstraDevice::AstraDevice(const string&)’:
/opt/ros/kinetic/include/ros/console.h:346:176: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 8 has type ‘boost::shared_ptr<openni::Device>::element_type* {aka openni::Device*}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/kinetic/include/ros/console.h:349:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(NULL, __VA_ARGS__)
     ^
/opt/ros/kinetic/include/ros/console.h:379:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/kinetic/include/ros/console.h:561:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/turtlebot/astra_ws/src/ros_astra_camera/src/astra_device.cpp:60:3: note: in expansion of macro ‘ROS_INFO’
   ROS_INFO("openni_device_: %x", openni_device_.get());
   ^
/opt/ros/kinetic/include/ros/console.h:346:176: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 8 has type ‘boost::shared_ptr<openni::Device>::element_type* {aka openni::Device*}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/kinetic/include/ros/console.h:349:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(NULL, __VA_ARGS__)
     ^
/opt/ros/kinetic/include/ros/console.h:379:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/kinetic/include/ros/console.h:561:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/turtlebot/astra_ws/src/ros_astra_camera/src/astra_device.cpp:62:3: note: in expansion of macro ‘ROS_INFO’
   ROS_INFO("openni_device_: %x, uri:%s", openni_device_.get(), device_URI.c_str());
   ^
/usr/bin/ld:/home/turtlebot/astra_ws/src/ros_astra_camera/include/openni2_redist/x64/libOpenNI2.so: file format not recognized; treating as linker script
/usr/bin/ld:/home/turtlebot/astra_ws/src/ros_astra_camera/include/openni2_redist/x64/libOpenNI2.so:1: syntax error
collect2: error: ld returned 1 exit status
make[2]: *** [/home/turtlebot/astra_ws/devel/.private/astra_camera/lib/libastra_wrapper.so] Error 1
make[1]: *** [CMakeFiles/astra_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/turtlebot/astra_ws/build/astra_camera; catkin build --get-env astra_camera | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................................
Failed     << astra_camera:make           [ Exited with code 2 ]      
Failed    <<< astra_camera                [ 1 minute and 8.1 seconds ]
[build] Summary: 0 of 1 packages succeeded.                           
[build]   Ignored:   None.                                            
[build]   Warnings:  None.                                            
[build]   Abandoned: None.                                            
[build]   Failed:    1 packages failed.                               
[build] Runtime: 1 minute and 8.2 seconds total.   
```
Not sure what to do

PhilipAmadasun, 2022-08-09:
@jian-dong  I get this error on catkin build now.
```
/usr/bin/ld:/home/turtlebot/astra_ws/src/ros_astra_camera/include/openni2_redist/x64/libOpenNI2.so: file format not recognized; treating as linker script
/usr/bin/ld:/home/turtlebot/astra_ws/src/ros_astra_camera/include/openni2_redist/x64/libOpenNI2.so:1: syntax error
collect2: error: ld returned 1 exit status
make[2]: *** [/home/turtlebot/astra_ws/devel/.private/astra_camera/lib/libastra_wrapper.so] Error 1
make[1]: *** [CMakeFiles/astra_wrapper.dir/all] Error 2
make: *** [all] Error 2
cd /home/turtlebot/astra_ws/build/astra_camera; catkin build --get-env astra_camera | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << astra_camera:make           [ Exited with code 2 ]               
Failed    <<< astra_camera                [ 2.0 seconds ]                      
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2.1 seconds total.  
```

jian-dong, 2022-08-16:
Hi @PhilipAmadasun , I remove LFS file.

closed 2022-08-16 by jian-dong