orbbec/ros_astra_camera #164: stream is not enabled

LeeSC427, 2022-10-04:
when I launched stereo_s_u3.launch, I got some warn messages (camera_params_template.yaml and launch file is not modified)

[ WARN] [1664868505.776232173]: Video mode Resolution :640x480@30Hz
format PIXEL_FORMAT_DEPTH_1_MM is not supported. 
[ WARN] [1664868505.776243499]: Default video mode Resolution :640x480@0Hz
format PIXEL_FORMAT_DEPTH_1_MMis not supported. Stream will be disabled.
[ WARN] [1664868505.776248226]: Supported video modes: 
[ WARN] [1664868505.776257908]: Resolution :320x200@15Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776262356]: Resolution :320x200@15Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776266415]: Resolution :320x200@30Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776270468]: Resolution :320x200@30Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776275290]: Resolution :320x200@60Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776280055]: Resolution :320x200@60Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776284753]: Resolution :640x400@15Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776290956]: Resolution :640x400@15Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776296136]: Resolution :640x400@30Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776300357]: Resolution :640x400@30Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776306010]: Resolution :640x400@60Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776310400]: Resolution :640x400@60Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776314841]: Resolution :1280x800@15Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776325680]: Resolution :1280x800@15Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776330637]: Resolution :1280x800@30Hz
format PIXEL_FORMAT_DEPTH_1_MM
[ WARN] [1664868505.776335341]: Resolution :1280x800@30Hz
format PIXEL_FORMAT_DEPTH_100_UM
[ WARN] [1664868505.776343132]: Video mode Resolution :640x480@30Hz
format PIXEL_FORMAT_GRAY16 is not supported. 
[ WARN] [1664868505.776348684]: Default video mode Resolution :640x480@0Hz
format PIXEL_FORMAT_GRAY16is not supported. Stream will be disabled.
[ WARN] [1664868505.776353879]: Supported video modes: 
[ WARN] [1664868505.776359031]: Resolution :320x200@15Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776364179]: Resolution :320x200@15Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776369222]: Resolution :320x200@15Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.776373897]: Resolution :320x200@30Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776378713]: Resolution :320x200@30Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776383589]: Resolution :320x200@30Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.776388536]: Resolution :320x200@60Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776394934]: Resolution :320x200@60Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776399438]: Resolution :320x200@60Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.776403515]: Resolution :640x400@15Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776407505]: Resolution :640x400@15Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776412355]: Resolution :640x400@15Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.776416563]: Resolution :640x400@30Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776420670]: Resolution :640x400@30Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776424876]: Resolution :640x400@30Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.776430149]: Resolution :640x400@60Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776436632]: Resolution :640x400@60Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776442451]: Resolution :640x400@60Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.776448175]: Resolution :1280x800@15Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776453703]: Resolution :1280x800@15Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776459621]: Resolution :1280x800@15Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.776465443]: Resolution :1280x800@30Hz
format PIXEL_FORMAT_GRAY16
[ WARN] [1664868505.776471375]: Resolution :1280x800@30Hz
format PIXEL_FORMAT_GRAY8
[ WARN] [1664868505.776477421]: Resolution :1280x800@30Hz
format PIXEL_FORMAT_RGB888
[ WARN] [1664868505.809041627]: Publishing dynamic camera transforms (/tf) at 10 Hz
[ERROR] [1664868505.809772615]: Tried to advertise a service that is already advertised in this node [/camera/camera/set_camera_info]
[ INFO] [1664868505.817993180]: OBCameraNode initialized
[ INFO] [1664868505.818008233]: Release device semaphore
[ INFO] [1664868505.818014620]: Release device semaphore done
[ INFO] [1664868505.818018951]: All devices connected
[ INFO] [1664868505.819153958]: init Done
[ INFO] [1664868505.819167060]: Creating camera node done...
[ INFO] [1664868521.574677123]: Image stream depth subscribed
[ WARN] [1664868521.574698013]: Stream depth is not enabled.
[ INFO] [1664868522.925677337]: Image stream depth unsubscribed

how should I solve these problems?

jian-dong, 2022-10-05:
Changing the resolution to 640x400 maybe can resolve this problem.

LeeSC427, 2022-10-05:
it works, thank you

LeeSC427, 2022-10-05:
it works, thanks

closed 2022-10-05 by LeeSC427