orbbec/ros_astra_camera #170: cannot calibrate color Astra mini camera

han88, 2023-01-06:
hi if I try to calibrate my Astra mini camera from the color stream /camera/color/image_raw with the ros calibration pkg the saved calibration file is depth_serialnumber. yaml.. if I change the name to rgb_serialnumber. yaml it's not recognized... I launch the astra mini with the Astra. launch file...
what do I have to change?
thanks in advance

jian-dong, 2023-01-09:
I've updated the readme. I hope that makes it clear.
- Set camera info uri, Go to `xxx.launch` 
```xml

<launch>
    <!--...-->
    <arg name="ir_info_uri" default="file:///you_ir_camera_calib_path/depth_camera.yaml"/>
    <arg name="color_info_uri" default="file:///you_depth_camera_calib_path/rgb_camera.yaml"/>
    <!--...-->
</launch>
```
- calibration file should like
```yaml
image_width: 640
image_height: 480
# The camera name is fixed. The color camera is rgb_camera, the depth/IR camera is depth_camera
camera_name: astra_rgb
camera_matrix:
  rows: 3
  cols: 3
  data: [517.301, 0, 326.785, 0, 519.291, 244.563, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.41527, 0.31874, -0.00197, 0.00071, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [0.999973, 0.00612598, -0.00406652, -0.00610201, 0.999964, 0.00588094, 0.0041024, -0.00585596, 0.999974 ]
projection_matrix:
  rows: 3
  cols: 4
  data: [517.301, 0, 326.785, -25.3167, 0, 519.291, 244.563, 0.282065, 0, 0, 1, 0.0777703]

```

han88, 2023-01-10:
thank you, i had to use ir_camera and rgb_camera as camera_name instead ...

closed 2023-01-10 by han88