orbbec/ros_astra_camera #172: astra pro无法获取红外图像和xyzrgb点云

guanba666, 2023-01-13:
使用最新的v1.2.1ros驱动
1、在astra_pro.launch文件将enable_point_cloud_xyzrgb设置为true，提示depth image frame id[] doesn't match RGB image frame id[]，该如何修改。
2、将enable_ir设置为true，没有接收到红外图像，如何获取红外图像。
3、将enable_color设置为false后，程序崩溃。

jian-dong, 2023-01-13:
@guanba666 你好，Astra pro改过了，您看一下

guanba666, 2023-01-13:
感谢，已成功运行，还请教一个问题：
运行rostopic echo /camera/extrinsic/depth_to_color时，该外参是astra自定义格式，ros无法识别该格式（cannot load message class for [astra_camera/Extrinsics]），我在代码将该参数打印出来，发现rotation和transition的值均为NAN，这是什么原因呢

guanba666, 2023-01-13:
另外/camear/color/camera_info，/camear/depth/camera_info，/camear/ir/camera_info这几个camera_info的值完全一样，这是正常的吗

jian-dong, 2023-01-13:
@guanba666 上面那个应该还是bug，我再看看，
“
另外/camear/color/camera_info，/camear/depth/camera_info，/camear/ir/camera_info这几个camera_info的值完全一样，这是正常的吗
“
IR和深度的相机参数是一样的，开了对齐是一样的

Hezhexi2002, 2023-01-28:
> @guanba666 上面那个应该还是bug，我再看看， “ 另外/camear/color/camera_info，/camear/depth/camera_info，/camear/ir/camera_info这几个camera_info的值完全一样，这是正常的吗 “ IR和深度的相机参数是一样的，开了对齐是一样的

您好，请问这个bug解决了吗，我现在运行`rostopic echo /camera/extrinsic/depth_to_color`提示：
[image]
另外通过`rosservice call /camera/get_camera_params "{}"`获取相机参数时则会有如下报错：
[image]
我使用的是最新的v1.2.2版本驱动，系统是Ubuntu20.04，ros版本为noetic

jian-dong, 2023-01-29:
@Hezhexi2002 您好，第一个问题，我看了Astra pro 相机固件里面本身没有内参，所以没法获取到，另外发布内参我做了配置在用参数`enable_publish_extrinsic`去开关，第二个问题是因为 `GetCameraParams`是自定义的type， 需要source `astra_camera`这个包的环境才能读取到这个类型。

Hezhexi2002, 2023-01-29:
> @Hezhexi2002 您好，第一个问题，我看了Astra pro 相机固件里面本身没有内参，所以没法获取到，另外发布内参我做了配置在用参数`enable_publish_extrinsic`去开关，第二个问题是因为 `GetCameraParams`是自定义的type， 需要source `astra_camera`这个包的环境才能读取到这个类型。

哦哦,谢谢大佬详细的回复,所以使用`rostopic echo /camera/extrinsic/depth_to_color`获取相机内参时首先需要在launch文件中将`enable_publish_extrinsic`参数设为true是吧,然后第二个问题source `astra_camera`这个包的环境具体是怎么操作呢,就是README中最开始的`source ./devel/setup.bash `操作吗,最后就是关于大佬您提到的Astra pro 相机固件里面本身没有内参,意思是需要我在launch文件中指定calibration file吗?

jian-dong, 2023-01-29:
@Hezhexi2002 您好，现在输出的内参是hard code上去的，如果需要准确的内参需要标定写到文件里面。source当前的环境就是编译好了 `source ./devel/setup.bash`就好了，`rostopic echo /camera/extrinsic/depth_to_color`这个topic应该输出的是深度到彩色的旋转和平移，属于外参，内参的话是 `rostopic echo /camera/depth/camera_info`

guanba666, 2023-01-29:
您好，在1.2.2版本中rostopic echo /camera/extrinsic/depth_to_color这个外参输出的值仍为nan

guanba666, 2023-01-29:
> @Hezhexi2002 您好，第一个问题，我看了Astra pro 相机固件里面本身没有内参，所以没法获取到，另外发布内参我做了配置在用参数`enable_publish_extrinsic`去开关，第二个问题是因为 `GetCameraParams`是自定义的type， 需要source `astra_camera`这个包的环境才能读取到这个类型。

这里《Astra pro 相机固件里面本身没有内参》是指没有/camera/extrinsic/depth_to_color这个外参吗，目前1.2.2版本depth_to_color输出的值仍为nan

Hezhexi2002, 2023-01-29:
> @Hezhexi2002 您好，现在输出的内参是hard code上去的，如果需要准确的内参需要标定写到文件里面。source当前的环境就是编译好了 `source ./devel/setup.bash`就好了，`rostopic echo /camera/extrinsic/depth_to_color`这个topic应该输出的是深度到彩色的旋转和平移，属于外参，内参的话是 `rostopic echo /camera/depth/camera_info`

好的,我明白了,但是现在我在launch文件中将`enable_publish_extrinsic`设置为true后再次运行`rostopic echo /camera/extrinsic/depth_to_color`输出仍然有warning看不到外参：
[image]
[image]
不过`rostopic echo /camera/depth/camera_info`可以获取到内参：
[image]
所以为什么现在还是无法获取到外参？另外还有就是我按照你们官方文档中安装相机驱动时`lsusb`命令检测不到相应设备：
[image]
但是使用dmesg | grep usb 查看内核装载 usb 设备提示信息确实又有相应的设备：
[image]
而且我运行文档中的NiViewer例程也没有问题：
[image]
所以我想问一下为什么`lsusb`会检测不到设备，是因为我没有配置udev规则吗，但是我在安装ros驱动时已经运行了相应的脚本，希望大佬能解答！

jian-dong, 2023-01-29:
> > @Hezhexi2002 您好，第一个问题，我看了Astra pro 相机固件里面本身没有内参，所以没法获取到，另外发布内参我做了配置在用参数`enable_publish_extrinsic`去开关，第二个问题是因为 `GetCameraParams`是自定义的type， 需要source `astra_camera`这个包的环境才能读取到这个类型。
> 
> 这里《Astra pro 相机固件里面本身没有内参》是指没有/camera/extrinsic/depth_to_color这个外参吗，目前1.2.2版本depth_to_color输出的值仍为nan

你好，因为固件里面读出来是`nan`,所以您看到的值也是`nan`

jian-dong, 2023-01-29:
> 好的,我明白了,但是现在我在launch文件中将`enable_publish_extrinsic`设置为true后再次运行`rostopic echo /camera/extrinsic/depth_to_color`输出仍然有warning看不到外参： [image] [image] 不过`rostopic echo /camera/depth/camera_info`可以获取到内参： [...]

您好，从这里看到是检查到设备了的，这里看到有`2bc5`的PID就是Astra的设备

Hezhexi2002, 2023-01-29:
> 

啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊

jian-dong, 2023-01-29:
> > 
> 
> 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊

应该就是显示问题吧，重新插拔一下显示应该正常了

Hezhexi2002, 2023-01-29:
> > > 
> > 
> > 
> > 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊
> 
> 应该就是显示问题吧，重新插拔一下显示应该正常了

试过重新拔插了，而且还重新配置了rules规则，但是还是没变化：
[image]

jian-dong, 2023-01-29:
> > > > 
> > > 
> > > 
> > > 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊
> > 
> > 
> > 应该就是显示问题吧，重新插拔一下显示应该正常了
> 
> 试过重新拔插了，而且还重新配置了rules规则，但是还是没变化： [image]

能够正常启动相机不？

Hezhexi2002, 2023-01-29:
> > > > > 
> > > > 
> > > > 
> > > > 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊
> > > 
> > > 
> > > 应该就是显示问题吧，重新插拔一下显示应该正常了
> > 
> > 
> > 试过重新拔插了，而且还重新配置了rules规则，但是还是没变化： [image]
> 
> 能够正常启动相机不？

可以正常运行文档中的NiViewer例程,前面我发了截图

jian-dong, 2023-01-29:
> > > > > > 
> > > > > 
> > > > > 
> > > > > 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊
> > > > 
> > > > 
> > > > 应该就是显示问题吧，重新插拔一下显示应该正常了
> > > 
> > > 
> > > 试过重新拔插了，而且还重新配置了rules规则，但是还是没变化： [image]
> > 
> > 
> > 能够正常启动相机不？
> 
> 可以正常运行文档中的NiViewer例程,前面我发了截图

`roslaunch astra_camera astro_pro.launch` 可以正常启动么。

Hezhexi2002, 2023-01-29:
> > > > > > > 
> > > > > > 
> > > > > > 
> > > > > > 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊
> > > > > 
> > > > > 
> > > > > 应该就是显示问题吧，重新插拔一下显示应该正常了
> > > > 
> > > > 
> > > > 试过重新拔插了，而且还重新配置了rules规则，但是还是没变化： [image]
> > > 
> > > 
> > > 能够正常启动相机不？
> > 
> > 
> > 可以正常运行文档中的NiViewer例程,前面我发了截图
> 
> `roslaunch astra_camera astro_pro.launch` 可以正常启动么。

可以,在rviz中查看rgb,depth以及ir图片都没有问题

guanba666, 2023-01-29:
> > > > > > 
> > > > > 
> > > > > 
> > > > > 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊
> > > > 
> > > > 
> > > > 应该就是显示问题吧，重新插拔一下显示应该正常了
> > > 
> > > 
> > > 试过重新拔插了，而且还重新配置了rules规则，但是还是没变化： [image]
> > 
> > 
> > 能够正常启动相机不？
> 
> 可以正常运行文档中的NiViewer例程,前面我发了截图,但是昨晚运行rtapmap的时候一直检测不到device: [image] 但是我在Detection/Select source里面已经选中了Orbbec astra pro了

你是用的main分支的驱动吗，我运行main分支也是no device detected，使用1.2.2版本是正常的

Hezhexi2002, 2023-01-29:
> 你是用的main分支的驱动吗，我运行main分支也是no device detected，使用1.2.2版本是正常的

是,我使用的是main分支

guanba666, 2023-01-29:
> 
不好意思，没跑过rtab-map

guanba666, 2023-01-29:
> > > > > > > > 
> > > > > > > 
> > > > > > > 
> > > > > > > 啊这，那个`2bc5`的PID的设备是我的鼠标和无线接收器啊，我之前在win下wsl2上`lsusb`PID为2bc5的设备确实是astra pro，但是后面是输出了astra pro的设备信息的，这里PID为2bc5的设备输出的是我鼠标和无线接收器的信息啊
> > > > > > 
> > > > > > 
> > > > > > 应该就是显示问题吧，重新插拔一下显示应该正常了
> > > > > 
> > > > > 
> > > > > 试过重新拔插了，而且还重新配置了rules规则，但是还是没变化： [image]
> > > > 
> > > > 
> > > > 能够正常启动相机不？
> > > 
> > > 
> > > 可以正常运行文档中的NiViewer例程,前面我发了截图,但是昨晚运行rtapmap的时候一直检测不到device: [image] 但是我在Detection/Select source里面已经选中了Orbbec astra pro了
> > 
> > 
> > 你是用的main分支的驱动吗，我运行main分支也是no device detected，使用1.2.2版本是正常的
> 
> 是,我使用的是main分支,意思大佬你用1.2.2版本跑rtabmap没问题?

不好意思，没跑过rtab-map

Hezhexi2002, 2023-01-29:
> no device detected

啊这,那大佬你说的no device detected是跑什么的时候遇见的呢?

jian-dong, 2023-01-29:
> > no device detected
> 
> 啊这,那大佬你说的no device detected是跑什么的时候遇见的呢?

@Hezhexi2002 我也没跑过rtabmap，你在x64还是ARM上运行的，我有空研究一下这个怎么跑的。

Hezhexi2002, 2023-01-29:
> > > no device detected
> > 
> > 
> > 啊这,那大佬你说的no device detected是跑什么的时候遇见的呢?
> 
> @Hezhexi2002 我也没跑过rtabmap，你在x64还是ARM上运行的，我有空研究一下这个怎么跑的。

我就是在intel nuc上跑的，系统是ubuntu20.04，ros版本是noetic，架构是x64

Hezhexi2002, 2023-01-30:
> > > no device detected
> > 
> > 
> > 啊这,那大佬你说的no device detected是跑什么的时候遇见的呢?
> 
> @Hezhexi2002 我也没跑过rtabmap，你在x64还是ARM上运行的，我有空研究一下这个怎么跑的。

今天我又重新试了一下将我的鼠标和无线连接器断开后再`lsusb`：
[image]
第一条`lsusb`命令是还未断开时候的输出，可以看到PID为2bc5的设备后面输出的还是我的鼠标和无线接收器的名称，第二条`lsusb`命令则是我断开鼠标和无线接收器之后的输出，可以看到PID为2bc5的设备后面变成了空白，我在想这是否是因为我udev的rules中设备名称配置出错的原因，但是我都是按照你们的文档安装的rules，之前在wsl2上运行都没问题，不知道为啥到了ubuntu下就不行了

78748547, 2023-02-07:
> 你好，因为固件里面读出来是`nan`,所以您看到的值也是`nan`

我也是相同的问题，我想请教一下大佬，这个深度相机和rgb相机中间隔开了好几厘米，为什么读出来的外参会是nan呢，这需要怎么修改呢？

jian-dong, 2023-02-07:
> 我也是相同的问题，我想请教一下大佬，这个深度相机和rgb相机中间隔开了好几厘米，为什么读出来的外参会是nan呢，这需要怎么修改呢？

您好，这是因为固件里面没写标定参数。

78748547, 2023-02-07:
> 您好，这是因为固件里面没写标定参数。

那请问一下如果不去标定外参，但是按照readme里面的方法在launch里面开启了对齐，那请问出来的/colour/image_raw和/depth/image_raw，/depth/points 这些数据是对齐后的么，我看他们的fram_id都是在 colour_optical_fram下的

jian-dong, 2023-02-07:
> 那请问一下如果不去标定外参，但是按照readme里面的方法在launch里面开启了对齐，那请问出来的/colour/image_raw和/depth/image_raw，/depth/points 这些数据是对齐后的么，我看他们的fram_id都是在 colour_optical_fram下的

可以标定内参的，默认对齐到彩色坐标系下面，这样就不用关心外参了。

78748547, 2023-02-07:
> 可以标定内参的，默认对齐到彩色坐标系下面，这样就不用关心外参了。

好的，十分感谢

sun-rabbit, 2023-10-20:
> 好的，十分感谢

你好，请问你解决了吗，我现在用astro pro跑rtabmap，能出现点云，但是建图时候hz很低，无法实时建图

sun-rabbit, 2023-10-20:
> 看到
你好，请问你解决了吗，我现在用astro pro跑rtabmap，能出现点云，但是建图时候hz很低，无法实时建图，我的是jeston agx orin ，ubuntu20.04，ros2 foxy