orbbec/ros_astra_camera #176: astra pro相机提示 wait for device to be connected

78748547, 2023-02-07:
按照readme配置好后，在ubuntu虚拟机上运行astrapro.launch文件可以正常运行。相同的配置步骤在jatson nano上运行后，终端一直提示等待设备连接，libuvc也可以确认已经成功配置好，请教各位大佬怎么解决wait for device to be connected的问题。
[image]

xaxxontech, 2023-02-07:
I have the same repeated "wait for device to be connected" issue with astra pro and the latest Jan 30 2023 commit af550123ca18ec2f405669f5df448fa01b7caaf3

however I tried checkout an earlier commit of 3 weeks ago ec531e1b88be34c16e967a5e971076b8aea4585f
and it works fine.
(this is right before "replace SDK to 86-beta6" commit, which I suspect has the bug)

78748547, 2023-02-08:
> I have the same repeated "wait for device to be connected" issue with astra pro and the latest Jan 30 2023 commit [af55012](https://github.com/orbbec/ros_astra_camera/commit/af550123ca18ec2f405669f5df448fa01b7caaf3)
> 
> however I tried checkout an earlier commit of 3 weeks ago [ec531e1](https://github.com/orbbec/ros_astra_camera/commit/ec531e1b88be34c16e967a5e971076b8aea4585f) and it works fine. (this is right before "replace SDK to 86-beta6" commit, which I suspect has the bug)

我尝试你的解决方案，它非常有用，非常感谢你！I tried the method you mentioned. It is very useful！Thank you very much!

78748547, 2023-02-09:
> I have the same repeated "wait for device to be connected" issue with astra pro and the latest Jan 30 2023 commit [...]

> I have the same repeated "wait for device to be connected" issue with astra pro and the latest Jan 30 2023 commit [...]

您好，再次请教您一个问题。不知道您在使用[ec531e1](https://github.com/orbbec/ros_astra_camera/commit/ec531e1b88be34c16e967a5e971076b8aea4585f) 版本时是否遇到过以下现象：我今天成功运行这个版本的驱动后发现，在rviz中 /camera/depth/points点云图无法显示，我可以确保launch文件中enable_depth等选项已经全部设置为true了（因为我的camera/depth_registered/points话题可以正常显示）。我echo了 /camera/depth/points话题，看见里面好像是有数据的。想请问一下您是否遇见过这种现象？
[image]

xaxxontech, 2023-02-09:
no problem here -- rviz PointCloud2 tool displays data from topic `/camera/depth/points` after running `roslaunch astra_camera astra_pro.launch` (using same ec531e1 version)

78748547, 2023-02-15:
@jian-dong 你好，我今天在nano上 重新尝试了一下 fix_MX400_conneted_bug分支 [8093afb](https://github.com/orbbec/ros_astra_camera/commit/8093afbed6130cb3fa8133c026b2f027c17af292)，但是发现它好像并没有作用，依然显示wait for device to be connected。

78748547, 2023-02-21:
> @jian-dong 你好，我今天在nano上 重新尝试了一下 fix_MX400_conneted_bug分支 [8093afb](https://github.com/orbbec/ros_astra_camera/commit/8093afbed6130cb3fa8133c026b2f027c17af292)，但是发现它好像并没有作用，依然显示wait for device to be connected。

今天我再次在nano上尝试了相同的fix_MX400_conneted_bug分支，结果他可以正常运行了。在此期间我什么也没有更改，只是删除了原来的包，并重新catkin_make一遍。现在问题解决了，感谢以上朋友的帮助。