orbbec/ros_astra_camera #186: jetson nano Installing error

KIYOLY, 2023-04-24:
> catkin_make --pkg astra_camera
`-- +++ processing catkin package: 'astra_camera'
-- ==> add_subdirectory(ros_astra_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found OpenCV: /usr (found version "4.1.1")
-- Boost version: 1.65.1
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for one of the modules 'libuvc'
-- Checking for one of the modules 'libglog'
-- ORRBEC Machine : aarch64

-- ORRBEC Machine Bits : 64

-- ORRBEC : arm64
-- libuvc ..
-- astra_camera: 3 messages, 8 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kim/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/kim/catkin_ws/build/ros_astra_camera"
####
Scanning dependencies of target geometry_msgs_generate_messages_py
Scanning dependencies of target std_srvs_generate_messages_py
Scanning dependencies of target std_srvs_generate_messages_nodejs
Scanning dependencies of target rosgraph_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_srvs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_lisp
Scanning dependencies of target std_srvs_generate_messages_cpp
Scanning dependencies of target actionlib_msgs_generate_messages_cpp
Scanning dependencies of target rosgraph_msgs_generate_messages_eus
Scanning dependencies of target geometry_msgs_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
Scanning dependencies of target roscpp_generate_messages_cpp
Scanning dependencies of target sensor_msgs_generate_messages_nodejs
Scanning dependencies of target actionlib_generate_messages_cpp
Scanning dependencies of target rosgraph_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_generate_messages_lisp
Scanning dependencies of target dynamic_reconfigure_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_generate_messages_nodejs
Scanning dependencies of target sensor_msgs_generate_messages_py
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_py
Scanning dependencies of target sensor_msgs_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_lisp
Scanning dependencies of target actionlib_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
Scanning dependencies of target std_srvs_generate_messages_lisp
Scanning dependencies of target std_msgs_generate_messages_py
Scanning dependencies of target roscpp_generate_messages_py
Scanning dependencies of target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[... 111 lines cut ...]
[ 34%] Generating C++ code from astra_camera/GetInt32.srv
[ 35%] Generating EusLisp code from astra_camera/GetCameraParams.srv
[ 36%] Generating Python code from SRV astra_camera/GetCameraInfo
[ 38%] Generating EusLisp code from astra_camera/GetString.srv
[ 39%] Generating Python msg __init__.py for astra_camera
[ 40%] Generating EusLisp code from astra_camera/SetInt32.srv
[ 42%] Generating EusLisp code from astra_camera/GetInt32.srv
[ 43%] Generating Python srv __init__.py for astra_camera
CMakeFiles/Makefile2:2967: recipe for target 'ros_astra_camera/CMakeFiles/cleanup_shm_node.dir/all' failed
make[1]: *** [ros_astra_camera/CMakeFiles/cleanup_shm_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 44%] Generating C++ code from astra_camera/SetString.srv
[ 46%] Generating C++ code from astra_camera/GetCameraInfo.srv
[ 47%] Generating EusLisp code from astra_camera/SetString.srv
[ 48%] Generating EusLisp code from astra_camera/GetCameraInfo.srv
[ 48%] Built target astra_camera_generate_messages_py
[ 50%] Generating EusLisp manifest code for astra_camera
[ 50%] Built target astra_camera_generate_messages_cpp
[ 50%] Built target astra_camera_generate_messages_eus
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

I'm using jetpack 4.6.1 and melodic
why this happens?

KIYOLY, 2023-04-24:
if i use
`catkin_make --pkg astra_camera -j1`

```
**make[2]:` *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/kim/catkin_ws/devel/lib/libastra_camera.so'.  Stop.

CMakeFiles/Makefile2:2016: recipe for target 'ros_astra_camera/CMakeFiles/astra_camera.dir/all' failed
make[1]: *** [ros_astra_camera/CMakeFiles/astra_camera.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed**
```

this error

ceoifung, 2023-06-27:
```shell
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/kim/catkin_ws/devel/lib/astra_camera/cleanup_shm_node'.  Stop.
make[2]: *** Waiting for unfinished jobs....
```
You should install `libopencv_objdetect.so.3.2.0` first, use apt install

```shell
sudo apt install libopencv3.2
```

ceoifung, 2023-06-27:
> `$ catkin_make --pkg astra_camera`
>
> ```
> -- +++ processing catkin package: 'astra_camera'
> -- ==> add_subdirectory(ros_astra_camera)
> -- Using these message generators: [...]

```shell
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/kim/catkin_ws/devel/lib/astra_camera/cleanup_shm_node'.  Stop.
make[2]: *** Waiting for unfinished jobs....
```
You should install libopencv_objdetect.so.3.2.0 first, use apt install
```
sudo apt install libopencv3.2
```